galexrt/dellhw_exporter #8: Add raid layout for vdisk

zoonage, 2017-11-01:
use case: 
Sometimes we don't care about a broken disk because of the a specific RAID setup, it would be nice for us to be able to use the RAID controller metric to alert on.

proposition:
I was thinking something like;

    dell_hw_storage_vdisk_raid{vdisk="0"} 10 // RAID 10
    dell_hw_storage_vdisk_raid{vdisk="1"} 5 // RAID 5

galexrt, 2017-11-01:
@zoonage There is a way to check if there is a broken disk without knowing which RAID is used in the system. I send you the Prometheus query when I find it.
But yes having "more" metrics about more than just state would be good.

zoonage, 2017-11-01:
I was thinking more along the lines of alerting rules, I'll start work on it and submit a pull request

galexrt, 2017-11-01:
@zoonage Thanks! With Prometheus query I meant an alert rule for alerting on failed/broken disks.

zoonage, 2017-11-01:
^ badly explained

There might be cases where if there were X failed disks then that would be fine in a particular raid setup but not another one

galexrt, 2017-11-01:
I understand now. Yes, a metric like `dell_hw_storage_vdisk_raid` would be useful.

If you need help with that, please let me know!

closed 2021-09-12 by galexrt

galexrt, 2021-09-12:
@zoonage Better late than never, a metric named `dell_hw_storage_vdisk_raidlevel` has been added. The new metrics exposes the **VDisk RAID level** as a metric.

commit b4697682 (2021-09-12): omreport: add vdisk raid level metric

This adds `dell_hw_storage_vdisk_raidlevel` metric, which holds the RAID
level of the VDISK.
Additionally the controller ID label was added to some metrics missing
it.

Resolves #8