galexrt/dellhw_exporter #9: Alerts / Graphs

cleonte, 2017-12-16:
Hi, 

Thanks for this, looks very good! btw, do you have any alerts defined or graphana dashboards?

Cheers

galexrt, 2017-12-17:
@cleonte I have to look into pushing example alerts and dashboards, as they are made by the company I currently work at. But it shouldn't be a problem of adding them.
I'll let you know when I have added them.

cleonte, 2017-12-17:
10q

At this point i'm using nagios with some home made scripts for dell :) but would like to  change to prometehus and you dell exporter si cool!

Cheers

altitude1326, 2018-01-30:
It will be very usefull if you can commit sample grafana dashboard. I’m not an expert but I could try to enrich it.

galexrt, 2018-02-07:
I finally had the time again, so I pushed a Grafana dashboard and a few Prometheus alert rules, see https://github.com/galexrt/dellhw_exporter/tree/dev/contrib/monitoring.
I will look into merging the changes on `dev` into `master` soon.

Thanks to @dhoeld for creating the Grafana dashboard and the Prometheus alerts initially and the company I work at!

closed 2018-02-07 by galexrt

cleonte, 2018-03-10:
sorry, so sorry, just saw your comment, will try it as soon as i can!

the alerts look cool! cant' wait to try the dashboard!