galexrt/dellhw_exporter #10: gometalinter issues

mjtrangoni, 2017-12-29:
Hi @galexrt ,

You could be interested seeing the output of [gometalinter](https://github.com/alecthomas/gometalinter), and you could even build a Makefile target running it automatically after fixing or excluding some issues.

See, 
`$ gometalinter --vendor ./...`

My 2 cents.

galexrt, 2017-12-30:
@mjtrangoni I'm happy to take PRs.
I'm already running gometalinter during development. I know about the existing warnings about missing comments on exported functions.

galexrt, 2018-01-02:
Some "issues" have been fixed in the latest `v1.3.0` release.
Missing comments on exported function are still present but will be added to at least the `pkg/omreport` when it is moved out to it's own repository soon.

galexrt, 2018-02-07:
I'm going to close this issue now, as every exported function in this repo has a comment now.

closed 2018-02-07 by galexrt