galexrt/dellhw_exporter #14: Predictive failure

cristianleontecsp, 2018-10-13:
Hi All,

how about a metric for predictive failure of the disks?

Cheers

galexrt, 2018-10-13:
@cristianleontecsp Does Dell OMSA have this functionality?

neuro42, 2021-08-24:
Yes.  Specifically, in StoragePdisk, fields[9] will be a "Yes" or "No" value indicating that the disk is predicted to fail.  I'm not sure if you want to add a new overall value for that (3?) or return the same value as non-critical (2)

Failure Predicted               : No

galexrt, 2021-08-29:
So if that info is provided by `omreport` I could probably add that as an additional metric somehow.

I'll check the coming days to check if that is easily doable.

galexrt, 2021-08-30:
New release 1.10.0 is pushed and currently being built, let me know if the new metric works well for you.

closed 2021-09-01 by galexrt

neuro42, 2021-09-08:
Just had a disk go to predictive failure which allowed this to be tested beyond a solid string of 0s.  Working as expected.

galexrt, 2021-09-08:
@neuro42 good to hear, thanks for letting me know!