galexrt/dellhw_exporter #15: which packages of OMSA should be use

alochym01, 2018-10-17:
Hi

when i un-tar  OM-SrvAdmin-Dell-Web-LX-9.1.0-2757.RHEL7.x86_64_A00.tar.gz, there are a lot of packages. 

Which one should be using for dellhw_exporter

Thank you

linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-itunnelprovider-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-rac4-populator-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-isvc-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-realssd-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-idracadm7-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-storage-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-cm-9.1.0-17.12.00.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-isvc-snmp-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-argtable2-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-sysfsutils-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-idracadm-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-racadm4-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-deng-snmp-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-smcommon-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-idrac-vmcli-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-rac-components-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-deng-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-omaksmig-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-xmlsup-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/libsmbios-2.3.1-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/raidcfg-6.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/syscfg-6.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-tomcat-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-racdrsc-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-idrac7-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-storage-snmp-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-omacs-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-storelib-sysfs-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-racadm5-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-racsvc-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-jre-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-idrac-snmp-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-hapi-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-storage-cli-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-idrac-ivmcli-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-omcommon-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-omacore-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-smweb-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-oslog-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-nvme-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-storelib-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-marvellib-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/smbios-utils-bin-2.3.1-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-ominst-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/srvadmin/RHEL7/x86_64/srvadmin-omilcore-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/metaRPMS/
linux/RPMS/supportRPMS/metaRPMS/RHEL7/
linux/RPMS/supportRPMS/metaRPMS/RHEL7/x86_64/
linux/RPMS/supportRPMS/metaRPMS/RHEL7/x86_64/srvadmin-storageservices-snmp-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/metaRPMS/RHEL7/x86_64/srvadmin-all-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/metaRPMS/RHEL7/x86_64/srvadmin-server-snmp-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/metaRPMS/RHEL7/x86_64/srvadmin-webserver-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/metaRPMS/RHEL7/x86_64/srvadmin-idrac-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/metaRPMS/RHEL7/x86_64/srvadmin-base-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/metaRPMS/RHEL7/x86_64/srvadmin-rac5-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/metaRPMS/RHEL7/x86_64/srvadmin-storageservices-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/metaRPMS/RHEL7/x86_64/srvadmin-standardAgent-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/metaRPMS/RHEL7/x86_64/srvadmin-storageservices-cli-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/metaRPMS/RHEL7/x86_64/srvadmin-server-cli-9.1.0-2757.12163.el7.x86_64.rpm
linux/RPMS/supportRPMS/metaRPMS/RHEL7/x86_64/srvadmin-rac4-9.1.0-2757.12163.el7.x86_64.rpm
linux/SRPMS/
linux/SRPMS/license.txt
[... 64 lines cut ...]
linux/custom/RHEL7/add-iDRAC/x86_64/srvadmin-idracadm-9.1.0-2757.12163.el7.x86_64.rpm
linux/custom/RHEL7/add-iDRAC/x86_64/srvadmin-racadm4-9.1.0-2757.12163.el7.x86_64.rpm
linux/custom/RHEL7/add-iDRAC/x86_64/srvadmin-idrac-vmcli-9.1.0-2757.12163.el7.x86_64.rpm
linux/custom/RHEL7/add-iDRAC/x86_64/srvadmin-rac-components-9.1.0-2757.12163.el7.x86_64.rpm
linux/custom/RHEL7/add-iDRAC/x86_64/srvadmin-racdrsc-9.1.0-2757.12163.el7.x86_64.rpm
linux/custom/RHEL7/add-iDRAC/x86_64/srvadmin-idrac7-9.1.0-2757.12163.el7.x86_64.rpm
linux/custom/RHEL7/add-iDRAC/x86_64/srvadmin-idrac-snmp-9.1.0-2757.12163.el7.x86_64.rpm
linux/custom/RHEL7/add-StorageManagement/x86_64/srvadmin-realssd-9.1.0-2757.12163.el7.x86_64.rpm
linux/custom/RHEL7/add-StorageManagement/x86_64/srvadmin-storage-9.1.0-2757.12163.el7.x86_64.rpm
linux/custom/RHEL7/add-StorageManagement/x86_64/srvadmin-sysfsutils-9.1.0-2757.12163.el7.x86_64.rpm
linux/custom/RHEL7/add-StorageManagement/x86_64/srvadmin-storage-snmp-9.1.0-2757.12163.el7.x86_64.rpm
linux/custom/RHEL7/add-StorageManagement/x86_64/srvadmin-storelib-sysfs-9.1.0-2757.12163.el7.x86_64.rpm
linux/custom/RHEL7/add-StorageManagement/x86_64/srvadmin-storage-cli-9.1.0-2757.12163.el7.x86_64.rpm
linux/custom/RHEL7/add-StorageManagement/x86_64/srvadmin-nvme-9.1.0-2757.12163.el7.x86_64.rpm
linux/custom/RHEL7/add-StorageManagement/x86_64/srvadmin-storelib-9.1.0-2757.12163.el7.x86_64.rpm
linux/custom/RHEL7/SA-WebServer/x86_64/srvadmin-smcommon-9.1.0-2757.12163.el7.x86_64.rpm
linux/custom/RHEL7/SA-WebServer/x86_64/srvadmin-omaksmig-9.1.0-2757.12163.el7.x86_64.rpm
linux/custom/RHEL7/SA-WebServer/x86_64/libsmbios-2.3.1-2757.12163.el7.x86_64.rpm
linux/custom/RHEL7/SA-WebServer/x86_64/srvadmin-tomcat-9.1.0-2757.12163.el7.x86_64.rpm
linux/custom/RHEL7/SA-WebServer/x86_64/srvadmin-omacs-9.1.0-2757.12163.el7.x86_64.rpm
linux/custom/RHEL7/SA-WebServer/x86_64/srvadmin-jre-9.1.0-2757.12163.el7.x86_64.rpm
linux/custom/RHEL7/SA-WebServer/x86_64/srvadmin-omcommon-9.1.0-2757.12163.el7.x86_64.rpm
linux/custom/RHEL7/SA-WebServer/x86_64/srvadmin-smweb-9.1.0-2757.12163.el7.x86_64.rpm
linux/custom/RHEL7/SA-WebServer/x86_64/smbios-utils-bin-2.3.1-2757.12163.el7.x86_64.rpm
linux/custom/RHEL7/SA-WebServer/x86_64/srvadmin-omilcore-9.1.0-2757.12163.el7.x86_64.rpm

galexrt, 2018-10-17:
Look at the Dockerfile for info what OMSA parts are installed: https://github.com/galexrt/dellhw_exporter/blob/master/Dockerfile#L13-L17

alochym01, 2018-10-17:
got it,

Thank you

closed 2018-10-17 by alochym01