galexrt/dellhw_exporter #16: how to install dellhw_exporter

alochym01, 2018-10-17:
hi

there is no section for how to install the 'dellhw_exporter'

i am a normal user, i dont know how to install 'dellhw_exporter'

please guide me how to install 'dellhw_export' to real server

galexrt, 2018-10-17:
You download a release binary for your architecture of dellhw_exporter and put it either in your path (recommended) or "somewhere else" and start it simply by starting it.
An example systemd unit file can be found here: https://github.com/galexrt/dellhw_exporter/tree/master/systemd

You may need to set the `omreport` binary path to the one on your system by using the `-collectors.omr-report` flag (https://github.com/galexrt/dellhw_exporter#flags).

alochym01, 2018-10-18:
its work properly

thank you

closed 2018-10-19 by alochym01