galexrt/dellhw_exporter #18: docker container not starting up: exit status 252

brintal, 2019-02-11:
I'm trying to run the docker image using the command: "docker run -d --name dellhw_exporter --privileged -p 9137:9137 galexrt/dellhw_exporter"

Unfortunately the container failes to start up. The log says:

```
time="2019-02-11T00:33:15Z" level=info msg="Starting srcds_exporter (version=1.3.2, branch=HEAD, revision=28563bef56c1c9ed7b2ac6ed4352706e5ca8e596)"
time="2019-02-11T00:33:15Z" level=info msg="Build context (go=go1.10.3, user=root@60ae3f9111bb, date=20180911-07:53:51)"
time="2019-02-11T00:33:15Z" level=info msg="Starting srvadmin-services ..."
time="2019-02-11T00:33:17Z" level=fatal msg="exit status 252"
```

galexrt, 2019-02-12:
@brintal Can you please try to run the container with the `--container=true` flag, to see if that fixes the issue

brintal, 2019-02-12:
@galexrt 
Thank you for your answer. I'm unsure what you mean. I'm not aware of any container flag in docker and the docker run help doesn't have any information about it either.

`
root@xxx:~# docker run -d --name dellhw_exporter --privileged --container=true -p 9137:9137 galexrt/dellhw_exporter
unknown flag: --container
See 'docker run --help'.
`
when i try "docker run -d --name dellhw_exporter --privileged -p 9137:9137 galexrt/dellhw_exporter --container=true" I get again the exit status 252.

galexrt, 2019-02-12:
@brintal Yes, the second one was meant.
Which architecture is your Docker host? Which OS is your Docker host running as?

brintal, 2019-02-12:
@galexrt 
I'm running Ubuntu Server 18.04 on a Dell PowerEdge R710. I think the problem is that I'm running OMSA also inside a docker container and therefore do not have access to the omreport command on my host. (should have read the readme properly).

galexrt, 2019-02-14:
@brintal So after stopping the other OMSA services and then running the dellhw_exporter, it worked?

brintal, 2019-05-29:
@galexrt 
Sorry for not getting back sooner. Actually it still doesn't work, even after installing omsa on the host system. I can run omreport successfully on the host and can get all the data I need from it.
But when I try to run the dellhw_exporter container, I still end up getting the exit status 252.

Because I'm not sure what exactly the container is doing, I tried to stop the OMSA services before running the container:
```
/opt/dell/srvadmin/sbin/srvadmin-services.sh stop
```
But that doesn't help. If the OMSA services are running or not, I always get the 252 error. 
Do you have any other idea?

galexrt, 2019-05-29:
@brintal What `docker run` command did you run right now?

brintal, 2019-05-29:
@galexrt 
I tried
```
docker run -d --name dellhw_exporter --privileged -p 9137:9137 galexrt/dellhw_exporter --container=true
```
and
````
docker run -d --name dellhw_exporter --privileged -p 9137:9137 galexrt/dellhw_exporter
```

galexrt, 2019-05-29:
@brintal Can you start the OMSA services on the server directly and then try running the `docker run` with `--container=false` for the flag of dellhw_exporter.

Let me know if that works and returns metrics.

brintal, 2019-05-29:
@galexrt 
[image]

container log:
```
time="2019-05-29T15:13:33Z" level=info msg="Starting srcds_exporter (version=1.3.2, branch=HEAD, revision=28563bef56c1c9ed7b2ac6ed4352706e5ca8e596)"
time="2019-05-29T15:13:33Z" level=info msg="Build context (go=go1.10.3, user=root@60ae3f9111bb, date=20180911-07:53:51)"
time="2019-05-29T15:13:33Z" level=info msg="Enabled collectors:"
time="2019-05-29T15:13:33Z" level=info msg=" - processors"
time="2019-05-29T15:13:33Z" level=info msg=" - storage_battery"
time="2019-05-29T15:13:33Z" level=info msg=" - volts"
time="2019-05-29T15:13:33Z" level=info msg=" - storage_pdisk"
time="2019-05-29T15:13:33Z" level=info msg=" - storage_vdisk"
time="2019-05-29T15:13:33Z" level=info msg=" - chassis"
time="2019-05-29T15:13:33Z" level=info msg=" - nics"
time="2019-05-29T15:13:33Z" level=info msg=" - ps"
time="2019-05-29T15:13:33Z" level=info msg=" - chassis_batteries"
time="2019-05-29T15:13:33Z" level=info msg=" - fans"
time="2019-05-29T15:13:33Z" level=info msg=" - ps_amps_sysboard_pwr"
time="2019-05-29T15:13:33Z" level=info msg=" - system"
time="2019-05-29T15:13:33Z" level=info msg=" - temps"
time="2019-05-29T15:13:33Z" level=info msg=" - memory"
time="2019-05-29T15:13:33Z" level=info msg=" - storage_controller"
time="2019-05-29T15:13:33Z" level=info msg=" - storage_enclosure"
```
looks good! I'll try to configure prometheus and report back if everything works. 
Thanks alot!

galexrt, 2019-05-29:
@brintal You can also just run `curl http://127.0.0.1:9137/metrics` and check if the `dellhw_probe_success` (something like that) is `1` or so + metrics are shown for the hardware.

brintal, 2019-05-29:
@galexrt 
Thanks alot for the hint. It seems like the collector is running but hardware metrics are still missing.
Output for `curl http://127.0.0.1:9137/metrics`:
```
# HELP dell_hw_scrape_collector_duration_seconds srcds_exporter: Duration of a collector scrape.
# TYPE dell_hw_scrape_collector_duration_seconds gauge
dell_hw_scrape_collector_duration_seconds{collector="chassis"} 1.029377127
dell_hw_scrape_collector_duration_seconds{collector="chassis_batteries"} 1.637599669
dell_hw_scrape_collector_duration_seconds{collector="fans"} 1.028994585
dell_hw_scrape_collector_duration_seconds{collector="memory"} 1.6199278750000001
dell_hw_scrape_collector_duration_seconds{collector="nics"} 1.625086019
dell_hw_scrape_collector_duration_seconds{collector="processors"} 1.046612909
dell_hw_scrape_collector_duration_seconds{collector="ps"} 1.028621878
dell_hw_scrape_collector_duration_seconds{collector="ps_amps_sysboard_pwr"} 1.625261183
dell_hw_scrape_collector_duration_seconds{collector="storage_battery"} 1.027360895
dell_hw_scrape_collector_duration_seconds{collector="storage_controller"} 1.029987998
dell_hw_scrape_collector_duration_seconds{collector="storage_enclosure"} 1.038245644
dell_hw_scrape_collector_duration_seconds{collector="storage_pdisk"} 1.032554272
dell_hw_scrape_collector_duration_seconds{collector="storage_vdisk"} 1.03420971
dell_hw_scrape_collector_duration_seconds{collector="system"} 1.038168414
dell_hw_scrape_collector_duration_seconds{collector="temps"} 1.038172124
dell_hw_scrape_collector_duration_seconds{collector="volts"} 1.028912679
# HELP dell_hw_scrape_collector_success srcds_exporter: Whether a collector succeeded.
# TYPE dell_hw_scrape_collector_success gauge
dell_hw_scrape_collector_success{collector="chassis"} 1
dell_hw_scrape_collector_success{collector="chassis_batteries"} 1
dell_hw_scrape_collector_success{collector="fans"} 1
dell_hw_scrape_collector_success{collector="memory"} 1
dell_hw_scrape_collector_success{collector="nics"} 1
dell_hw_scrape_collector_success{collector="processors"} 1
dell_hw_scrape_collector_success{collector="ps"} 1
dell_hw_scrape_collector_success{collector="ps_amps_sysboard_pwr"} 1
dell_hw_scrape_collector_success{collector="storage_battery"} 1
dell_hw_scrape_collector_success{collector="storage_controller"} 1
dell_hw_scrape_collector_success{collector="storage_enclosure"} 1
dell_hw_scrape_collector_success{collector="storage_pdisk"} 1
dell_hw_scrape_collector_success{collector="storage_vdisk"} 1
dell_hw_scrape_collector_success{collector="system"} 1
dell_hw_scrape_collector_success{collector="temps"} 1
dell_hw_scrape_collector_success{collector="volts"} 1
# HELP go_gc_duration_seconds A summary of the GC invocation durations.
# TYPE go_gc_duration_seconds summary
go_gc_duration_seconds{quantile="0"} 0.000147265
go_gc_duration_seconds{quantile="0.25"} 0.000234384
go_gc_duration_seconds{quantile="0.5"} 0.000271096
go_gc_duration_seconds{quantile="0.75"} 0.000300482
go_gc_duration_seconds{quantile="1"} 0.000518752
go_gc_duration_seconds_sum 0.034282861
go_gc_duration_seconds_count 126
# HELP go_goroutines Number of goroutines that currently exist.
# TYPE go_goroutines gauge
go_goroutines 59
# HELP go_memstats_alloc_bytes Number of bytes allocated and still in use.
# TYPE go_memstats_alloc_bytes gauge
go_memstats_alloc_bytes 2.259384e+06
# HELP go_memstats_alloc_bytes_total Total number of bytes allocated, even if freed.
# TYPE go_memstats_alloc_bytes_total counter
go_memstats_alloc_bytes_total 1.45766304e+08
# HELP go_memstats_buck_hash_sys_bytes Number of bytes used by the profiling bucket hash table.
# TYPE go_memstats_buck_hash_sys_bytes gauge
go_memstats_buck_hash_sys_bytes 1.454539e+06
# HELP go_memstats_frees_total Total number of frees.
# TYPE go_memstats_frees_total counter
go_memstats_frees_total 447108
# HELP go_memstats_gc_sys_bytes Number of bytes used for garbage collection system metadata.
# TYPE go_memstats_gc_sys_bytes gauge
go_memstats_gc_sys_bytes 512000
# HELP go_memstats_heap_alloc_bytes Number of heap bytes allocated and still in use.
# TYPE go_memstats_heap_alloc_bytes gauge
go_memstats_heap_alloc_bytes 2.259384e+06
[... 48 lines cut ...]
# HELP go_memstats_sys_bytes Number of bytes obtained by system. Sum of all system allocations.
# TYPE go_memstats_sys_bytes gauge
go_memstats_sys_bytes 1.5902968e+07
# HELP process_cpu_seconds_total Total user and system CPU time spent in seconds.
# TYPE process_cpu_seconds_total counter
process_cpu_seconds_total 3.46
# HELP process_max_fds Maximum number of open file descriptors.
# TYPE process_max_fds gauge
process_max_fds 1.048576e+06
# HELP process_open_fds Number of open file descriptors.
# TYPE process_open_fds gauge
process_open_fds 75
# HELP process_resident_memory_bytes Resident memory size in bytes.
# TYPE process_resident_memory_bytes gauge
process_resident_memory_bytes 9.371648e+06
# HELP process_start_time_seconds Start time of the process since unix epoch in seconds.
# TYPE process_start_time_seconds gauge
process_start_time_seconds 1.55914281266e+09
# HELP process_virtual_memory_bytes Virtual memory size in bytes.
# TYPE process_virtual_memory_bytes gauge
process_virtual_memory_bytes 2.2802432e+07
```
there are no errors in the container log but if I try to call omreport from inside the container, I get:
[image]
Same call on the host gives me detailed information about all physical disks.

galexrt, 2019-05-29:
@brintal Could you run the `omreport storage pdisk controller=0` command again but like this: `omreport storage pdisk contorller=0; echo "ret: $?"`.

When I'm at home I have to use the weekend to begin setting up my rack again to be able to test it.

brintal, 2019-05-29:
@galexrt 
[image]

Thank you so much. Of course I would be happy to get some more support with this issue but really don't worry too much about it.

galexrt, 2019-05-29:
@brintal I have raised #20 to fix the `dell_hw_scrape_collector_success` metrics. I'll merge it soon and look into making a new release too.

galexrt, 2019-05-29:
@brintal Could you please try the latest `master` tag (run `docker pull galexrt/dellhw_exporter:master` and then the `docker run`) with the `--container=false` flag. Please check the `curl http://127.0.0.1:9137/metrics` if the `dell_hw_scrape_collector_success` also show `0` for you now.

Thanks!

brintal, 2019-05-29:
@galexrt 
```
docker run -d --name dellhw_exporter --privileged -p 9137:9137 galexrt/dellhw_exporter:master --container=false

Unable to find image 'galexrt/dellhw_exporter:master' locally
master: Pulling from galexrt/dellhw_exporter
8ba884070f61: Pull complete
9669a56a01c8: Pull complete
c5d88204ee51: Pull complete
Digest: sha256:589be7cabb7f58a7c42d72968a902693f3f4af1a038b6849d375fe88792af9f9
Status: Downloaded newer image for galexrt/dellhw_exporter:master
4c7e66dfe6a04aa6a4cd08546baa85d6cd73067a6620cb5d1dc57c58f93c8708
```
container log:
```
time="2019-05-29T16:47:44Z" level=info msg="Starting srcds_exporter (version=1.3.2, branch=master, revision=15db03c289feca5aace6fbf452441703a19c00f4)"
time="2019-05-29T16:47:44Z" level=info msg="Build context (go=go1.12.5, user=root@9b03577b3753, date=20190529-16:35:58)"
time="2019-05-29T16:47:44Z" level=info msg="Enabled collectors:"
time="2019-05-29T16:47:44Z" level=info msg=" - chassis_batteries"
time="2019-05-29T16:47:44Z" level=info msg=" - nics"
time="2019-05-29T16:47:44Z" level=info msg=" - chassis"
time="2019-05-29T16:47:44Z" level=info msg=" - ps_amps_sysboard_pwr"
time="2019-05-29T16:47:44Z" level=info msg=" - storage_controller"
time="2019-05-29T16:47:44Z" level=info msg=" - storage_vdisk"
time="2019-05-29T16:47:44Z" level=info msg=" - system"
time="2019-05-29T16:47:44Z" level=info msg=" - fans"
time="2019-05-29T16:47:44Z" level=info msg=" - processors"
time="2019-05-29T16:47:44Z" level=info msg=" - storage_battery"
time="2019-05-29T16:47:44Z" level=info msg=" - storage_pdisk"
time="2019-05-29T16:47:44Z" level=info msg=" - temps"
time="2019-05-29T16:47:44Z" level=info msg=" - volts"
time="2019-05-29T16:47:44Z" level=info msg=" - memory"
time="2019-05-29T16:47:44Z" level=info msg=" - ps"
time="2019-05-29T16:47:44Z" level=info msg=" - storage_enclosure"
time="2019-05-29T16:47:50Z" level=error msg="volts collector failed after 1.021282s: exit status 255"
time="2019-05-29T16:47:50Z" level=error msg="processors collector failed after 1.021829s: exit status 255"
time="2019-05-29T16:47:50Z" level=error msg="ps collector failed after 1.024451s: exit status 255"
time="2019-05-29T16:47:50Z" level=error msg="temps collector failed after 1.026193s: exit status 255"
time="2019-05-29T16:47:50Z" level=error msg="storage_pdisk collector failed after 1.027203s: exit status 7"
time="2019-05-29T16:47:50Z" level=error msg="storage_controller collector failed after 1.043739s: exit status 7"
time="2019-05-29T16:47:50Z" level=error msg="fans collector failed after 1.052270s: exit status 255"
time="2019-05-29T16:47:50Z" level=error msg="chassis_batteries collector failed after 1.451200s: exit status 255"
time="2019-05-29T16:47:51Z" level=error msg="nics collector failed after 1.635093s: exit status 255"
time="2019-05-29T16:48:29Z" level=error msg="fans collector failed after 1.020769s: exit status 255"
time="2019-05-29T16:48:29Z" level=error msg="storage_controller collector failed after 1.025716s: exit status 7"
time="2019-05-29T16:48:29Z" level=error msg="temps collector failed after 1.029556s: exit status 255"
time="2019-05-29T16:48:29Z" level=error msg="volts collector failed after 1.033294s: exit status 255"
time="2019-05-29T16:48:29Z" level=error msg="ps collector failed after 1.033677s: exit status 255"
time="2019-05-29T16:48:29Z" level=error msg="processors collector failed after 1.033147s: exit status 255"
time="2019-05-29T16:48:29Z" level=error msg="storage_pdisk collector failed after 1.033621s: exit status 7"
time="2019-05-29T16:48:30Z" level=error msg="nics collector failed after 1.667449s: exit status 255"
time="2019-05-29T16:48:30Z" level=error msg="chassis_batteries collector failed after 1.679966s: exit status 255"
```

`curl http://127.0.0.1:9137/metrics:`
````
# HELP dell_hw_scrape_collector_duration_seconds srcds_exporter: Duration of a collector scrape.
# TYPE dell_hw_scrape_collector_duration_seconds gauge
dell_hw_scrape_collector_duration_seconds{collector="chassis"} 1.035492338
dell_hw_scrape_collector_duration_seconds{collector="chassis_batteries"} 1.679966296
dell_hw_scrape_collector_duration_seconds{collector="fans"} 1.020768839
dell_hw_scrape_collector_duration_seconds{collector="memory"} 1.685189373
dell_hw_scrape_collector_duration_seconds{collector="nics"} 1.6674490180000001
dell_hw_scrape_collector_duration_seconds{collector="processors"} 1.03314726
dell_hw_scrape_collector_duration_seconds{collector="ps"} 1.033676943
dell_hw_scrape_collector_duration_seconds{collector="ps_amps_sysboard_pwr"} 1.68427264
dell_hw_scrape_collector_duration_seconds{collector="storage_battery"} 1.02554927
dell_hw_scrape_collector_duration_seconds{collector="storage_controller"} 1.025715506
dell_hw_scrape_collector_duration_seconds{collector="storage_enclosure"} 1.033287206
dell_hw_scrape_collector_duration_seconds{collector="storage_pdisk"} 1.033621196
[... 98 lines cut ...]
# TYPE go_memstats_stack_sys_bytes gauge
go_memstats_stack_sys_bytes 1.736704e+06
# HELP go_memstats_sys_bytes Number of bytes obtained by system. Sum of all system allocations.
# TYPE go_memstats_sys_bytes gauge
go_memstats_sys_bytes 7.2679672e+07
# HELP process_cpu_seconds_total Total user and system CPU time spent in seconds.
# TYPE process_cpu_seconds_total counter
process_cpu_seconds_total 0.06
# HELP process_max_fds Maximum number of open file descriptors.
# TYPE process_max_fds gauge
process_max_fds 1.048576e+06
# HELP process_open_fds Number of open file descriptors.
[... cut ...]
process_open_fds 64
# HELP process_resident_memory_bytes Resident memory size in bytes.
# TYPE process_resident_memory_bytes gauge
process_resident_memory_bytes 7.094272e+06
# HELP process_start_time_seconds Start time of the process since unix epoch in seconds.
# TYPE process_start_time_seconds gauge
process_start_time_seconds 1.55914846335e+09
# HELP process_virtual_memory_bytes Virtual memory size in bytes.
# TYPE process_virtual_memory_bytes gauge
process_virtual_memory_bytes 1.1497472e+08

```

galexrt, 2019-05-29:
Interesting that some metric collectors (e.g., `omreport chassis`) are still reporting to be successful.

I'll look into that more when I have my lab setup up'n'running.

Thanks for the provided outputs!

***

Coming back to the original issue with the OMSA service being unable to be started in the container through the dellhw_exporter.

Could you look into the OMSA / srvadmin log files on the host after you have run `/opt/dell/srvadmin/sbin/srvadmin-services.sh start` to see if it works.

Also the containerized version starts the srvadm-services in the container by default, could you try the following steps:
```
/opt/dell/srvadmin/sbin/srvadmin-services.sh stop
ps aux | grep srvadmin
# should result in no processes running anymore
docker run -d --name dellhw_exporter --privileged -p 9137:9137 galexrt/dellhw_exporter:master --container=true
docker logs -f dellhw_exporter
# is it working now or still failing?
```

brintal, 2019-05-29:
@galexrt 

I'm not sure which log is the correct one. This is from `/opt/dell/srvadmin/var/log/openmanage/dsm_om_connsvcdIO.log` after calling ´/opt/dell/srvadmin/sbin/srvadmin-services.sh start`.
If you need any other log please let me know. 
```
29-May-2019 20:26:33.556 WARNING [WebServerThread] org.apache.tomcat.util.digester.SetPropertiesRule.begin [SetPropertiesRule]{Server/Service/Engine/Host/Context} Setting property 'debug' to '0' did not find a matching property.
29-May-2019 20:26:33.594 INFO [WebServerThread] org.apache.catalina.startup.VersionLoggerListener.log Server version:        Apache Tomcat/8.0.21
29-May-2019 20:26:33.594 INFO [WebServerThread] org.apache.catalina.startup.VersionLoggerListener.log Server built:          Mar 23 2015 14:11:21 UTC
29-May-2019 20:26:33.594 INFO [WebServerThread] org.apache.catalina.startup.VersionLoggerListener.log Server number:         8.0.21.0
29-May-2019 20:26:33.594 INFO [WebServerThread] org.apache.catalina.startup.VersionLoggerListener.log OS Name:               Linux
29-May-2019 20:26:33.595 INFO [WebServerThread] org.apache.catalina.startup.VersionLoggerListener.log OS Version:            4.15.0-50-generic
29-May-2019 20:26:33.595 INFO [WebServerThread] org.apache.catalina.startup.VersionLoggerListener.log Architecture:          amd64
29-May-2019 20:26:33.595 INFO [WebServerThread] org.apache.catalina.startup.VersionLoggerListener.log Java Home:             /opt/dell/srvadmin/lib64/openmanage/jre
29-May-2019 20:26:33.595 INFO [WebServerThread] org.apache.catalina.startup.VersionLoggerListener.log JVM Version:           1.8.0_74-b02
29-May-2019 20:26:33.595 INFO [WebServerThread] org.apache.catalina.startup.VersionLoggerListener.log JVM Vendor:            Oracle Corporation
29-May-2019 20:26:33.596 INFO [WebServerThread] org.apache.catalina.startup.VersionLoggerListener.log CATALINA_BASE:         /opt/dell/srvadmin/lib64/openmanage/apache-tomcat
29-May-2019 20:26:33.596 INFO [WebServerThread] org.apache.catalina.startup.VersionLoggerListener.log CATALINA_HOME:         /opt/dell/srvadmin/lib64/openmanage/apache-tomcat
29-May-2019 20:26:33.596 INFO [WebServerThread] org.apache.catalina.startup.VersionLoggerListener.log Command line argument: -Djava.library.path=�:/opt/dell/srvadmin//bin:/opt/dell/srvadmin//sbin:/opt/dell/srvadmin//lib64:/opt/dell/srvadmin//lib64/openmanage:/opt/dell/srvadmin//lib64/openmanage/jre/lib/i386:/opt/dell/srvadmin//lib64/openmanage/jre/lib/i386/client:/opt/dell/srvadmin//lib64/openmanage/jre/lib/amd64:/opt/dell/srvadmin//lib64/openmanage/jre/lib/amd64/server:/opt/dell/srvadmin//lib64/openmanage/jre/bin
29-May-2019 20:26:33.596 INFO [WebServerThread] org.apache.catalina.startup.VersionLoggerListener.log Command line argument: -DSERVICE
29-May-2019 20:26:33.597 INFO [WebServerThread] org.apache.catalina.startup.VersionLoggerListener.log Command line argument: -Xms32m
29-May-2019 20:26:33.597 INFO [WebServerThread] org.apache.catalina.startup.VersionLoggerListener.log Command line argument: -Xmx256m
29-May-2019 20:26:33.597 INFO [WebServerThread] org.apache.catalina.startup.VersionLoggerListener.log Command line argument: -Dcatalina.home=/opt/dell/srvadmin//lib64/openmanage/apache-tomcat
29-May-2019 20:26:33.597 INFO [WebServerThread] org.apache.catalina.startup.VersionLoggerListener.log Command line argument: -Dcatalina.base=/opt/dell/srvadmin//lib64/openmanage/apache-tomcat
29-May-2019 20:26:33.598 INFO [WebServerThread] org.apache.catalina.startup.VersionLoggerListener.log Command line argument: -DCATALINA_HOME=/opt/dell/srvadmin//lib64/openmanage/apache-tomcat
29-May-2019 20:26:33.598 INFO [WebServerThread] org.apache.catalina.startup.VersionLoggerListener.log Command line argument: -DCATALINA_BASE=/opt/dell/srvadmin//lib64/openmanage/apache-tomcat
29-May-2019 20:26:33.598 INFO [WebServerThread] org.apache.catalina.startup.VersionLoggerListener.log Command line argument: -Djava.util.logging.manager=org.apache.juli.ClassLoaderLogManager
29-May-2019 20:26:33.598 INFO [WebServerThread] org.apache.catalina.startup.VersionLoggerListener.log Command line argument: -Djava.util.logging.config.file=/opt/dell/srvadmin//lib64/openmanage/apache-tomcat/conf/logging.properties
29-May-2019 20:26:33.844 INFO [WebServerThread] org.apache.coyote.AbstractProtocol.init Initializing ProtocolHandler ["http-nio-1311"]
29-May-2019 20:26:34.576 INFO [WebServerThread] org.apache.tomcat.util.net.NioSelectorPool.getSharedSelector Using a shared selector for servlet write/read
29-May-2019 20:26:34.580 INFO [WebServerThread] org.apache.catalina.startup.Catalina.load Initialization processed in 1831 ms
29-May-2019 20:26:34.622 INFO [WebServerThread] org.apache.catalina.core.StandardService.startInternal Starting service Catalina
29-May-2019 20:26:34.623 INFO [WebServerThread] org.apache.catalina.core.StandardEngine.startInternal Starting Servlet Engine: Apache Tomcat/8.0.21
29-May-2019 20:26:35.307 INFO [localhost-startStop-1] org.apache.jasper.servlet.TldScanner.scanJars At least one JAR was scanned for TLDs yet contained no TLDs. Enable debug logging for this logger for a complete list of JARs that were scanned but no TLDs were found in them. Skipping unneeded JARs during scanning can improve startup time and JSP compilation time.
29-May-2019 20:26:35.442 SEVERE [localhost-startStop-1] org.apache.catalina.session.PersistentManagerBase.startInternal No Store configured, persistence disabled
[... cut ...]
root@ceres:/opt/dell/srvadmin/var/log/openmanage# ps aux | grep srvadmin
root      7212  0.0  0.0  13136  1044 pts/0    S+   20:47   0:00 grep --color=auto srvadmin
root@ceres:/opt/dell/srvadmin/var/log/openmanage# ps aux | grep srvadmin
root      7221  0.0  0.0  13136  1004 pts/0    S+   20:48   0:00 grep --color=auto srvadmin
root@ceres:/opt/dell/srvadmin/var/log/openmanage# ps aux | grep srvadmin
root      7229  0.0  0.0  13136  1012 pts/0    S+   20:48   0:00 grep --color=auto srvadmin
root@ceres:/opt/dell/srvadmin/var/log/openmanage#
```

galexrt, 2019-06-05:
@brintal I have now removed the functionality behind `--container` flag in latest `master` and have switched to using an entrypoint script in the Docker image to start the services.

Testing with my trusty old Dell Poweredge 2950:

```
  Operating System: CentOS Linux 7 (Core)
       CPE OS Name: cpe:/o:centos:centos:7
            Kernel: Linux 3.10.0-957.el7.x86_64
      Architecture: x86-64
```

And I then ran the exporter in Docker with `docker run -d --name dellhw_exporter --privileged -p 9137:9137 quay.io/galexrt/dellhw_exporter:docker_entrypoint` (`docker_entrypoint` changes will soon be merged to `master`).

I had issues with Fedora, as Fedora uses 5.x kernel which does not seem to compatible with Dell OMSA tolls (kernel modules are failing to compile).

galexrt, 2020-05-25:
Closing as no response.

***

I haven't heard of people having this issue anymore and there have been quite a few changes to the Dockerfile and other code.

closed 2020-05-25 by galexrt

brintal, 2020-05-26:
@galexrt 
Sorry for not replying anymore. I just tried the latest tag and it seems to work fine on my R710. Thanks!