galexrt/dellhw_exporter #37: Releases: tarballs

HP41, 2020-03-23:
Seems like not all releases don't have tarballs in them.
Do you know if there's a plan to publish tarball releases in github as well.
We use them to deploy them via chef is why.

Additionally I noticed some discrepancies between release versions and actual binary versions. For eg: v1.3.5 and v1.3.2.

Thank you again!

HP41, 2020-03-23:
I noticed 1.4.3 has the tarball!

Feel free to comment more on the original comment or close the issue as needed!

Thank you!

galexrt, 2020-03-23:
There had been some issues due to CircleCI API changes which had already been resolved in v1.4.2 but that release didn't got published for some reason.

As written in my comment in #35, I have to get the ci / build stuff in-line and fixed for all of my Golang project soon / ASAP.

closed 2020-03-23 by galexrt