galexrt/dellhw_exporter #38: manual setup

andrewm659, 2020-04-14:
So i'm trying to set this up manually but I don't see what port it runs on or what I need to put in my prometheus configuration so ti scrapes it.  Can you help with this and can you update the README.md to provide this in the future?

andrewm659, 2020-04-14:
nm.

closed 2020-04-14 by andrewm659

galexrt, 2020-04-14:
I have added some documentation around running the dellhw_exporter without Docker and also an example static Prometheus job config to the `README.md` file.

andrewm659, 2020-04-14:
Also was this intended to run on a physical server running linux i'm guessing?

galexrt, 2020-04-16:
Yes, you can run it on a Linux server directly. The Docker image "packaging" is just for easy for people running, e.g., Docker and / or Kubernetes.

commit a8be1e47 (2020-04-14): readme: added more instructions to readme file

Resolve #38.