galexrt/dellhw_exporter #42: Minor trouble with updated build routines (promu-related)

sfudeus, 2020-06-09:
Hi Alex, 

while trying to use your updated build setup with `promu` I stumbled over some minor issues which break our builds. 

a) `promu` is referenced  as `$(PROMU)` in the Makefile, which is only set to `promu`, assuming that it lives in the path. I suggest to use either the location where you put it (`$(FIRST_GOPATH)/bin/promu`) or set that as variable `PROMU`

b) `Makefile` assumes `PREFIX` being set when calling `promu --prefix $(PREFIX)`, but `PREFIX` might not be set. I suggest to add `--prefix $(PREFIX)` only when `PREFIX` is set. 

Cheers,
Stephan

galexrt, 2020-06-09:
@sfudeus Hi Stephan,

1. Yeah, makes sense. I have switched it to use the downloaded promu binary path.
2. The `CWD` variable should have been used there, instead of the `PREFIX`.

Will push a new release in a bit that includes those fixes.

Please let me know if that resolves the issues found.

closed 2020-06-09 by galexrt

sfudeus, 2020-06-09:
thx, looks good so far

commit db84cbcd (2020-06-09): ci: fix build routine issues

Fix the reported issues with the `PREFIX` var (should be `CWD`) and
setting the downloaded promu as the `PROMU` variable instead of just
`promu`.

Resolves #42.