galexrt/dellhw_exporter #43: Support for firmware and bios versions (Feature request)

sfudeus, 2020-06-09:
`omreport` is capable of extracting the version for some firmware and BIOS - it would be really helpful to be able to extract them and expose them as metrics (as a label). 
This can help monitoring outdated firmwares or heterogenous cluster states. 

I have half a PR in the making for this.

galexrt, 2020-06-09:
@sfudeus That sounds great! Please let me know if you need any help with the code.

Out of curiosity, which `omreport` subcommand is reporting this information?

sfudeus, 2020-06-09:
`omreport chassis bios` and `omreport chassis firmware`

sfudeus, 2020-06-09:
`omreport chassis info` is another one for the drac version. Unfortunately, there are a bunch of more firmwares where I did not find the version (nic, backplane, raid, hdd, ...).

galexrt, 2020-06-09:
Closed by #44.

closed 2020-06-09 by galexrt