galexrt/dellhw_exporter #59: H750 RAID cannot obtain storage disk information, can upgrade to OMSA10.2.0.0 to fix it.

lovebaicai, 2021-12-10:
dell model：PowerEdge R740
raid model：H750
link: https://dl.dell.com/FOLDER07619118M/1/OM-SrvAdmin-Dell-Web-LX-10.2.0.0-4631_A00.tar.gz

galexrt, 2021-12-13:
Could you post the output of the `omreport storage controller` and then using the controller id `omreport storage pdisk controller=ID` on the host.

Which version of the dellhw_exporter are you running? Also which omreport version `omreport version`?

lovebaicai, 2021-12-14:
- Use raid H750, old version cannot get storage info. Contacted the engineer at dell, tell me I need to upgrade the omsa version to 10.2.0.0. The version installed by yum is not the latest.
- old omreport version
```
Product name   : Dell EMC OpenManage Server Administrator
Version        : 10.1.0.0
Copyright      : Copyright (C) Dell Inc. 1995-2021 All rights reserved.
Company        : Dell Inc.
Latest Version : 10.2.0.0
Download Link  : https://www.dell.com/support/home/us/en/04/drivers/driversdetails?driverId=4M1GP
```
- error log like this:
```
# omreport storage controller
No controllers found

# omreport storage pdisk controller=0
Invalid controller value. Read, controller=0
No controllers found.
```
- os version
```
Linux 3.10.0-1160.el7.x86_64 #1 SMP Mon Oct 19 16:18:59 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
CentOS Linux release 7.9.2009 (Core)
```

galexrt, 2021-12-14:
> The version installed by yum is not the latest.

Are you using the dellhw_exporter Container Image? Meaning that the omreport Version needs to be updated in the image.

lovebaicai, 2021-12-15:
Sure, i use dellhw_exporter  image for latest version, but this omreport version is not the latest. so in the end I used binary deployment.

galexrt, 2021-12-15:
The dellhw_exporter container image uses the DellHW Linux Repositories (https://linux.dell.com/repo/hardware/dsu/), but they don't seem to contain 10.2.x version (yet?).

When they release the new packages to the repo they will be picked up soon when a new dellhw_exporter version has been released.

lovebaicai, 2021-12-15:
ok，thanks you, this issue close.

closed 2021-12-15 by lovebaicai