galexrt/dellhw_exporter #68: Have a make rpm target?

YPatois, 2022-09-01:
Hi,

To deploy this package on RH derivates, a "make rpm" target could be useful.
I forked and edited a bit to make a crude (working) stuff, but that should be enhanced.

Is that something that could be useful to others?

Regards,

Yannick

galexrt, 2022-09-01:
I think some people might find it useful, though I don't have a repository server to host it.

Would it "suffice" to have the RPM package pushed as an artifact of dellhw_exporter GitHub releases?

YPatois, 2022-09-02:
Hi,
Thanks for your reply.
Having the RPM pushed is a first good enough step. Then people who can host it could wrok from that.

However, the way I edited the code make this process still complicated, as version is hardcoded, etc.

Using autotools or cmake  would makes it way easier, but I don't want to break everything that's working fine and done some way certainly for good reasons.

If you want, you can have a look at what I did and tell if that may be merged at some point.
Then, if you agree I would likely take a minimal approach, putting a dependency on autotools to generates a properly version tagged dellhw_exporter.spec from some dellhw_exporter.spec.in (alternative is to use a simple awk/whatever hook to grab current version and stuff it into the spec file) so that the RPM can be built from any tagged release just by a "make rpm"' command.

What do you think?

Tks.

galexrt, 2022-09-04:
> If you want, you can have a look at what I did and tell if that may be merged at some point.

A PR of the current state sounds good.

Using a script to "extract" the version number from the `VERSION` file would be good.

YPatois, 2022-09-04:
Hi,
Thanks.
Let me a week or so to clean it properly and I'll make a PR, using the VERSION file value as you suggested.
I'll let you know.

YPatois, 2022-09-14:
Hi,

A bit late but here where I am: I added a way to dynamically add the VERSION tag in the spec file (using awk - hum, should add that in the builddeps), however, I see some inconsistancy in the files and would like your opinion:
In Makefile, line 71 (your code, not my fork):
cp .build/dellhw_exporter $(PACKAGE_DIR)/usr/bin

In systemd/dellhw_exporter.service, line 7:
ExecStart=/usr/sbin/dellhw_exporter $OPTIONS
Packaging is done in bin directory, while the systemd calls from sbin.
I fixed that in RPM by moving dellhw_exporter to sbin, not bin.

But was the cp to bin intentional or a bug?

Thanks.

galexrt, 2022-09-14:
@YPatois No worries, thanks for working on this!

> But was the cp to bin intentional or a bug?

Should be a bug.

Feel free to throw all your changes into a PR and I'll gladly review them :)

galexrt, 2022-12-09:
Closing as we have a make rpm target now thanks to your PR #69.

closed 2022-12-09 by galexrt