galexrt/dellhw_exporter #70: How should run exporter to connect Dell EMC device

babsdn, 2022-12-09:
I have some devices from EMC (Dell ) now I did follow steps would you please help me :

 My Os that want run exporter on that is Redhat 8 :  
1-  curl -O https://linux.dell.com/repo/hardware/dsu/bootstrap.cgi
2. bash bootstrap.cgi
3. yum install srvadmin-all
I did above steps . What should I do for the next step ?
Where should insert username- password - ip address to connect dell device ?

galexrt, 2022-12-09:
You need to run the dellhw_exporter on each of your Dell EMC devices and scrape the dellhw_exporter on each machines using Prometheus.

babsdn, 2022-12-09:
> You need to run the dellhw_exporter on each of your Dell EMC devices and scrape the dellhw_exporter on each machines using Prometheus.

Thanks. But EMC devices such as Unity . we cannot run a script on them . That is not possible.

galexrt, 2022-12-13:
@babsdn Currently the dellhw_exporter can't be used for these cases as it relies on running the `omreport` command on the same node it is running on.

babsdn, 2022-12-13:
As I said previously I did follow steps :

1- curl -O https://linux.dell.com/repo/hardware/dsu/bootstrap.cgi
2. bash bootstrap.cgi
3. yum install srvadmin-all
I think last step is installing omreport . Is that correct ?
Please give me a clear answer can we use dellhw_exporter to use and monitor EMC devices ? Please explain more as you know we cannot install any packages directly on EMC devices.

galexrt, 2022-12-15:
@babsdn `omreport` should be part of the `srvadmin-all` package, you can test that by running `omreport` on one of the EMC devices on which you just installed the `srvadmin-all` package.

You need to install and run the `dellhw_exporter` binary on these EMC devices where you just installed the `srvadmin-all` package from Dell.

If you can't run any binaries on these EMC devices as written, the dellhw_exporter won't work for you as it is relying on being able to run `omreport` to get the metrics of the device it is running on.

babsdn, 2022-12-15:
I have a linux server "ip = 10.20.30.40" that omreport and srvadmin-all has been installed on that 
my emc unity devices ip address is 10.40.50.60 . Now How can run this exporter ? Where should define username - password for connect to emc devices ?

galexrt, 2022-12-15:
You can't use the dellhw_exporter to get metrics from an external device like that.

The dellhw_exporter must run on the same device, as the exporter currently can't retrieve the metrics from the web interface nor from "remote" devices due to the relience on the "local" `omreport` command.

babsdn, 2022-12-15:
That is strange . As you know better than me we cannot run omreport or dellexporter directly on EMC unity cli . it does not support

galexrt, 2022-12-15:
I have not worked with EMC devices nor the EMC unity cli.

As written this case is not supported with the dellhw_exporter it needs to be run directly on the server/device.

***

Closing as adding a way to remotely execute `omreport` or use the Dell srvadmin web interface/ API is out of scope for this project. If someone wants to contribute this as a feature feel free to open an issue for discussion/ pull request for code review, thanks!

closed 2022-12-15 by galexrt