galexrt/dellhw_exporter #93: storage_pdisk collector failed

AyoubH, 2024-01-25:
Hello

I am getting collectors errors failing, most of time it's fans and storage_pdisk collectors

/var/log/messages:146458:Jan 25 10:25:41 <myserver> dellhw_exporter[1020987]: time="<some time>" level=error msg="storage_pdisk collector failed after 0.388469s: exit status 255"

is there anyone who can help get rid of this messages and solve the issue?
Thanks

paulbarfuss, 2024-01-30:
+1

I am seeing the same error. It seems that if you don't have any disks on `omreport store controller=0` it errors out and doesn't check the remaining disks.

```
# omreport storage pdisk controller=0
List of Physical Disks on Controller Dell XXX XXX

Controller Dell XXX XXX
No Physical Disks found

# echo $?
255
```

In my case, I actually am interested in the Pdisks on controller=2 so perhaps just need to handle the `255` error code.

AyoubH, 2024-01-30:
It does see the disk, although it still errors out

`# omreport storage pdisk controller=0
....
ID                              : 0:1:0
Status                       : Ok
Name                        : Physical Disk 0:1:0
State                         : Ready
...`

galexrt, 2024-02-02:
@paulbarfuss @AyoubH could you please provide the entire output from a failing command and a successful storage pdisk command so I can analyze it and add tests for that case.

AyoubH, 2024-02-02:
@galexrt I would really like to thank you for this great work you have done and still doing for the community.

After installing the exporter in many servers, It works fine in some of them, no errors and even no warning, except for some where a random of bunch collectors are facing some errors, most of the time it's storage_pdisk, PS, and fan collectors.

**FAILING COMMAND OUTPUT** 
`In this server the dellhw exporter errors out dellhw_exporter: time="2024-01-29T04:58:31-05:00" level=error msg="storage_pdisk collector failed after 0.297482s: exit status 255`

'''shell [myserver111 - username]# omreport storage pdisk controller=0
List of Physical Disks on Controller 

Controller PERC H330 Mini (Embedded)
ID                              : 0:1:0
Status                          : Ok
Name                            : Physical Disk 0:1:0
State                           : Ready
Power Status                    : Not Applicable
Bus Protocol                    : SATA
Media                           : SSD
Part of Cache Pool              : Not Applicable
Remaining Rated Write Endurance : 100%
Failure Predicted               : No
Revision                        : XXXXXX
Driver Version                  : Not Applicable
Model Number                    : Not Applicable
T10 PI Capable                  : No
Certified                       : Yes
Encryption Capable              : No
Encryption Protocol             : Not Applicable
Encrypted                       : Not Applicable
Progress                        : Not Applicable
Mirror Set ID                   : Not Applicable
Capacity                        : 223.00 GB (239444426752 bytes)
Used RAID Disk Space            : 0.00 GB (0 bytes)
Available RAID Disk Space       : 223.00 GB (239444426752 bytes)
Hot Spare                       : No
Vendor ID                       : DELL(tm)
Product ID                      : XXXXXXXX
Serial No.                      : XXXXXXXXXXXXXXX
Part Number                     : XXXXXXXXXXXXXXXXXXXXX
Negotiated Speed                : 6.00 Gbps
Capable Speed                   : 6.00 Gbps
PCIe Negotiated Link Width      : Not Applicable
PCIe Maximum Link Width         : Not Applicable
Sector Size                     : 512B
Device Write Cache              : Not Applicable
Manufacture Day                 : Not Available
Manufacture Week                : Not Available
Manufacture Year                : Not Available
SAS Address                     : 0xXXXXXXXXXXXX
WWN                             : 0xXXXXXXXXXXXXX
Non-RAID HDD Disk Cache Policy  : Not Applicable
Disk Cache Policy               : Not Applicable
Sub Vendor                      : Not Available
Available Spare                 : Not Available
Cryptographic Erase Capable     : Yes
'''

**SUCCESSFUL COMMAND OUTPUT**
'''shell [myserver222 - username]# omreport storage pdisk controller=0
List of Physical Disks on Controller

Controller BOSS-S1 (Embedded)
ID                              : 0
Status                          : Ok
Name                            : Physical Disk 0
State                           : Online
Power Status                    : Not Applicable
Bus Protocol                    : SATA
Media                           : SSD
Part of Cache Pool              : Not Applicable
[... 38 lines cut ...]
**I'd like to capture your attention a different error bellow, in which storage_vdisk is concerned.**
`dellhw_exporter: time="2024-02-02T07:00:59-05:00" level=error msg="storage_vdisk collector failed after 0.150900s: strconv.ParseFloat: parsing \"\": invalid syntax"`

'''shell [myserver333 - username]# omreport storage vdisk
List of Virtual Disks in the System

Controller BOSS-S1 (Embedded)
ID                                : 0
Status                            : Ok
Name                              : VD_R1_1
State                             : Ready
Encrypted                         : Not Applicable
Layout                            : RAID-1
Size                              : 223.51 GB (239990276096 bytes)
T10 Protection Information Status : No
Associated Fluid Cache State      : Not Applicable
Device Name                       : Not Available
Bus Protocol                      : SATA
Media                             : SSD
Read Policy                       : No Read Ahead
Write Policy                      : Write Through
Cache Policy                      : Not Applicable
Stripe Element Size               : 64 KB
Disk Cache Policy                 : Disabled
'''

galexrt, 2024-02-11:
Thank you for the output and for pointing out another issue!

I forgot that the `omreport` parsing/executer logic uses the semicolon separate format which now makes it hard to just throw in the outputs into the unit tests and debug from my side.

Please provide the command outputs with the following `omreport` flags: `omreport -fmt ssv ...` and their respective exit codes when you ran them `echo $?`. Please include the mentioned issue outputs as well so I can add unit tests for these cases.

lchopfpt, 2024-02-12:
So on the node where "no virtual disks being shown":
`omreport storage vdisk -fmt ssv
List of Virtual Disks in the System

ID;Status;Name;State;Encrypted;Layout;Size;T10 Protection Information Status;Associated Fluid Cache State ;Device Name;Bus Protocol;Media;Read Policy;Write Policy;Cache Policy;Stripe Element Size;Disk Cache Policy`
Exit Code: 0

For the storage pdisk, the issue is that it's not finding any controllers so not passing any to "storage pdisk".  In this case, there does seem to be something funky on this host that is not the fault of this exporter.   So we can probably skip this being an issue with this exporter.

`# omreport storage pdisk -fmt ssv
Error! Required parameter missing: controller

# echo $?
255

 omreport storage controller -fmt ssv
 Controller  PCIe SSD Subsystem(Not Available)

Controller

# echo $?
0
`

galexrt, 2024-02-13:
I have released version 1.13. with some logging changes, I might be able to start pinpointing the issues mentioned here.

Please post the new logs of the dellhw_exporter which shows the error logs.

AyoubH, 2024-02-13:
Hello

I have installed the latest version 1.13.3
```shell root@myserver myuser # dellhw_exporter --version
dellhw_exporter, version 1.13.3 (branch: HEAD, revision: 88bea66b189c3be8660f2f2d6bc86cf6ba23bc26)
```
 **and below the output**
```shell root@myserver myuser # dellhw_exporter --log-level DEBUG
INFO[0000] Starting dellhw_exporter (version=1.13.3, branch=HEAD, revision=88bea66b189c3be8660f2f2d6bc86cf6ba23bc26)
INFO[0000] Build context (go=go1.18.10, platform=linux/amd64, user=root@6b35ba1b9776, date=20240213-13:27:14, tags=netgo static_build)
INFO[0000] Setting command timeout to 15
INFO[0000] Caching is disabled by default
INFO[0000] Enabled collectors:
INFO[0000]  - fans
INFO[0000]  - nics
INFO[0000]  - storage_enclosure
INFO[0000]  - storage_vdisk
INFO[0000]  - system
INFO[0000]  - temps
INFO[0000]  - volts
INFO[0000]  - chassis
INFO[0000]  - firmwares
INFO[0000]  - processors
INFO[0000]  - ps_amps_sysboard_pwr
INFO[0000]  - storage_battery
INFO[0000]  - ps
INFO[0000]  - storage_controller
INFO[0000]  - storage_pdisk
INFO[0000]  - chassis_batteries
INFO[0000]  - memory
DEBU[0001] Waiting for collectors
DEBU[0002] system collector succeeded after 0.086672s.
DEBU[0002] ps collector succeeded after 0.089446s.
DEBU[0002] temps collector succeeded after 0.094158s.
DEBU[0002] processors collector succeeded after 0.096242s.
DEBU[0002] fans collector succeeded after 0.104480s.
DEBU[0002] volts collector succeeded after 0.105054s.
DEBU[0002] storage_enclosure collector succeeded after 0.111973s.
DEBU[0002] storage_vdisk collector succeeded after 0.122696s.
DEBU[0002] storage_battery collector succeeded after 0.123452s.
DEBU[0002] firmwares collector succeeded after 0.284401s.
DEBU[0002] storage_controller collector succeeded after 0.342501s.
DEBU[0002] chassis_batteries collector succeeded after 0.351601s.
DEBU[0002] ps_amps_sysboard_pwr collector succeeded after 0.380746s.
DEBU[0002] nics collector succeeded after 0.390761s.
DEBU[0002] memory collector succeeded after 0.397066s.
ERRO[0002] storage_pdisk collector failed after 0.439935s: strconv.ParseFloat: parsing "": invalid syntax
```
**also**
```shell 
Feb 13 10:40:08 myserver dellhw_exporter: time="2024-02-13T10:40:08-05:00" level=error msg="storage_pdisk collector failed after 0.452978s: strconv.ParseFloat: parsing \"\": invalid syntax"
```

lchopfpt, 2024-02-13:
`level=info msg="Starting dellhw_exporter (version=1.13.3, branch=HEAD, revision=88bea66b189c3be8660f2f2d6bc86cf6ba23bc26)"`
I set -log-level=debug
Not really seeing any different logs?
`time="2024-02-13T08:46:15-08:00" level=info msg="Starting dellhw_exporter (version=1.13.3, branch=HEAD, revision=88bea66b189c3be8660f2f2d6bc86cf6ba23bc26)"
time="2024-02-13T08:46:15-08:00" level=info msg="Build context (go=go1.18.10, platform=linux/amd64, user=root@6b35ba1b9776, date=20240213-13:27:14, tags=netgo static_build)"
time="2024-02-13T08:46:15-08:00" level=info msg="Setting command timeout to 15"
time="2024-02-13T08:46:15-08:00" level=info msg="Caching is disabled by default"
time="2024-02-13T08:46:15-08:00" level=info msg="Enabled collectors:"
time="2024-02-13T08:46:15-08:00" level=info msg=" - volts"
time="2024-02-13T08:46:15-08:00" level=info msg=" - memory"
time="2024-02-13T08:46:15-08:00" level=info msg=" - storage_battery"
time="2024-02-13T08:46:15-08:00" level=info msg=" - storage_vdisk"
time="2024-02-13T08:46:15-08:00" level=info msg=" - nics"
time="2024-02-13T08:46:15-08:00" level=info msg=" - processors"
time="2024-02-13T08:46:15-08:00" level=info msg=" - storage_pdisk"
time="2024-02-13T08:46:15-08:00" level=info msg=" - chassis"
time="2024-02-13T08:46:15-08:00" level=info msg=" - chassis_batteries"
time="2024-02-13T08:46:15-08:00" level=info msg=" - firmwares"
time="2024-02-13T08:46:15-08:00" level=info msg=" - fans"
time="2024-02-13T08:46:15-08:00" level=info msg=" - temps"
time="2024-02-13T08:46:15-08:00" level=info msg=" - storage_enclosure"
time="2024-02-13T08:46:15-08:00" level=info msg=" - system"
time="2024-02-13T08:46:15-08:00" level=info msg=" - ps"
time="2024-02-13T08:46:15-08:00" level=info msg=" - ps_amps_sysboard_pwr"
time="2024-02-13T08:46:15-08:00" level=info msg=" - storage_controller"
time="2024-02-13T08:47:20-08:00" level=debug msg="Waiting for collectors"
time="2024-02-13T08:47:20-08:00" level=debug msg="temps collector succeeded after 0.086983s."
time="2024-02-13T08:47:20-08:00" level=debug msg="system collector succeeded after 0.094560s."
time="2024-02-13T08:47:20-08:00" level=debug msg="fans collector succeeded after 0.094713s."
time="2024-02-13T08:47:20-08:00" level=debug msg="ps collector succeeded after 0.097929s."
time="2024-02-13T08:47:20-08:00" level=debug msg="processors collector succeeded after 0.103702s."
time="2024-02-13T08:47:20-08:00" level=debug msg="storage_enclosure collector succeeded after 0.103935s."
time="2024-02-13T08:47:20-08:00" level=debug msg="volts collector succeeded after 0.114934s."
time="2024-02-13T08:47:20-08:00" level=debug msg="storage_battery collector succeeded after 0.128980s."
time="2024-02-13T08:47:20-08:00" level=error msg="storage_vdisk collector failed after 0.140017s: strconv.ParseFloat: parsing \"\": invalid syntax"
`

galexrt, 2024-02-13:
@lchopfpt I went ahead and started adding a logger to the collector logic, can you try with 1.3.4 https://github.com/galexrt/dellhw_exporter/releases/tag/v1.13.4. Please run the exporter with debug level

Could you please see if the log shows info now when the pdisk collector fails now

AyoubH, 2024-02-14:
@galexrt I tried with 1.13.4, the logs 

```shell
DEBU[0024] iterating pdisks from controller 0, data: [{Name:storage_pdisk_status Value:0 Labels:map[controller:0 controller_name:PERC H330 Mini (Embedded) disk:0_1_0]} {Name:storage_pdisk_state Value: Labels:map[controller:0 controller_name:PERC H330 Mini (Embedded) disk:0_1_0]} {Name:storage_pdisk_failure_predicted Value:0 Labels:map[controller:0 controller_name:PERC H330 Mini (Embedded) disk:0_1_0]} {Name:storage_pdisk_remaining_rated_write_endurance Value:100 Labels:map[controller:0 controller_name:PERC H330 Mini (Embedded) disk:0_1_0]}]  controller=0
ERRO[0024] storage_pdisk collector failed after 0.452719s: strconv.ParseFloat: parsing "": invalid syntax
```

lchopfpt, 2024-02-14:
I see pdisk debugging, but not vdisk debugging.
``` level=debug msg="iterating pdisks from controller 0, data: [{Name:storage_pdisk_status Value:0 Labels:map[controller:0 controller_name:PERC H730 Mini (Embedded) disk:0_1_0]} {Name:storage_pdisk_state Value:2 Labels:map[controller:0 controller_name:PERC H730 Mini (Embedded) disk:0_1_0]} {Name:storage_pdisk_failure_predicted Value:0 Labels:map[controller:0 controller_name:PERC H730 Mini (Embedded) disk:0_1_0]} {Name:storage_pdisk_remaining_rated_write_endurance Value:-1 Labels:map[controller:0 controller_name:PERC H730 Mini (Embedded) disk:0_1_0]} {Name:storage_pdisk_status Value:0 Labels:map[controller:0 controller_name:PERC H730 Mini (Embedded) disk:0_1_1]} {Name:storage_pdisk_state Value:2 Labels:map[controller:0 controller_name:PERC H730 Mini (Embedded) disk:0_1_1]} {Name:storage_pdisk_failure_predicted Value:0 Labels:map[controller:0 controller_name:PERC H730 Mini (Embedded) disk:0_1_1]} {Name:storage_pdisk_remaining_rated_write_endurance Value:-1 Labels:map[controller:0 controller_name:PERC H730 Mini (Embedded) disk:0_1_1]}]" controller=0
```

Will need the same debugging added for vdisk as well.  
```Feb 14 07:50:34 nomdb802 dellhw_exporter[2120901]: time="2024-02-14T07:50:34-08:00" level=error msg="storage_vdisk collector failed after 0.134843s: strconv.ParseFloat: parsing \"\": invalid syntax"```

galexrt, 2024-02-14:
@AyoubH can you please provide the output of `omreport storage pdisk -fmt ssv -controller=0` without it I can't figure why the `storage_pdisk_state` is not having a value set.

@lchopfpt I'm going to create `1.13.5` patch release tomorrow adding the vdisk debugging. I'll post an update when I have released that version.

AyoubH, 2024-02-15:
@galexrt here is the output

```shell
root@myserver myuser # omreport storage pdisk controller=0 -fmt ssv
List of Physical Disks on Controller PERC H330 Mini (Embedded)

Controller PERC H330 Mini (Embedded)

ID;Status;Name;State;Power Status;Bus Protocol;Media;Part of Cache Pool;Remaining Rated Write Endurance;Failure Predicted;Revision;Driver Version;Model Number;T10 PI Capable;Certified;Encryption Capable;Encryption Protocol;Encrypted;Progress;Mirror Set ID;Capacity;Used RAID Disk Space;Available RAID Disk Space;Hot Spare;Vendor ID;Product ID;Serial No.;Part Number;Negotiated Speed;Capable Speed;PCIe Negotiated Link Width;PCIe Maximum Link Width;Sector Size;Device Write Cache;Manufacture Day;Manufacture Week;Manufacture Year;SAS Address;WWN;Non-RAID HDD Disk Cache Policy;Disk Cache Policy;Sub Vendor;Available Spare;Cryptographic Erase Capable
0:1:0;Ok;Physical Disk 0:1:0;Non-RAID;Not Applicable;SATA;SSD;Not Applicable;100%;No;D1DF005;Not Applicable;Not Applicable;No;Yes;No;Not Applicable;Not Applicable;Not Applicable;Not Applicable;446.63 GB (479559942144 bytes);0.00 GB (0 bytes);446.63 GB (479559942144 bytes);No;DELL(tm);MTFDDAK480TDN;2014274E8D30;SG0D35F3MCS0004416JZA02;6.00 Gbps;6.00 Gbps;Not Applicable;Not Applicable;512B;Not Applicable;Not Available;Not Available;Not Available;0x4433221104000000;0x4433221104000000;Not Applicable;Not Applicable;Not Available;Not Available;Yes
```

AyoubH, 2024-02-15:
@galexrt I confirm that pdisk collector failing is fixed in the 1.13.5 version,
for the the vdisk collector, got the debug errors bellow from the node that has the issue

```shell
DEBU[0031] iterating vdisk values, data: {Name:storage_vdisk_status Value:0 Labels:map[controller_name:BOSS-S1 (Embedded) vdisk:0 vdisk_name:VD_R1_1]}  vdisk="{storage_vdisk_status 0 map[controller_name:BOSS-S1 (Embedded) vdisk:0 vdisk_name:VD_R1_1]}"
DEBU[0031] iterating vdisk values, data: {Name:storage_vdisk_state Value:1 Labels:map[controller_name:BOSS-S1 (Embedded) vdisk:0 vdisk_name:VD_R1_1]}  vdisk="{storage_vdisk_state 1 map[controller_name:BOSS-S1 (Embedded) vdisk:0 vdisk_name:VD_R1_1]}"
DEBU[0031] iterating vdisk values, data: {Name:storage_vdisk_raidlevel Value:111 Labels:map[controller_name:BOSS-S1 (Embedded) vdisk:0 vdisk_name:VD_R1_1]}  vdisk="{storage_vdisk_raidlevel 111 map[controller_name:BOSS-S1 (Embedded) vdisk:0 vdisk_name:VD_R1_1]}"
DEBU[0031] iterating vdisk values, data: {Name:storage_vdisk_read_policy Value: Labels:map[controller_name:BOSS-S1 (Embedded) vdisk:0 vdisk_name:VD_R1_1]}  vdisk="{storage_vdisk_read_policy  map[controller_name:BOSS-S1 (Embedded) vdisk:0 vdisk_name:VD_R1_1]}"
ERRO[0031] storage_vdisk collector failed after 0.392404s: strconv.ParseFloat: parsing "": invalid syntax
```

the output of the omreport storage vdisk command:
```shell
root@server1 user1 # omreport storage vdisk
List of Virtual Disks in the System

Controller BOSS-S1 (Embedded)
ID                                : 0
Status                            : Ok
Name                              : VD_R1_1
State                             : Ready
Encrypted                         : Not Applicable
Layout                            : RAID-1
Size                              : 111.73 GB (119966990336 bytes)
T10 Protection Information Status : No
Associated Fluid Cache State      : Not Applicable
Device Name                       : Not Available
Bus Protocol                      : SATA
Media                             : SSD
Read Policy                       : No Read Ahead
Write Policy                      : Write Through
Cache Policy                      : Not Applicable
Stripe Element Size               : 64 KB
Disk Cache Policy                 : Disabled
```

lchopfpt, 2024-02-15:
My vdisk output:
`time="2024-02-15T08:32:56-08:00" level=debug msg="iterating vdisk values, data: {Name:storage_vdisk_status Value:0 Labels:map[controller_name:PERC H730 Mini (Embedded) vdisk:0 vdisk_name:Virtual Disk0]}" vdisk="{storage_vdisk_status 0 map[controller_name:PERC H730 Mini (Embedded) vdisk:0 vdisk_name:Virtual Disk0]}"
time="2024-02-15T08:32:56-08:00" level=debug msg="iterating vdisk values, data: {Name:storage_vdisk_state Value:1 Labels:map[controller_name:PERC H730 Mini (Embedded) vdisk:0 vdisk_name:Virtual Disk0]}" vdisk="{storage_vdisk_state 1 map[controller_name:PERC H730 Mini (Embedded) vdisk:0 vdisk_name:Virtual Disk0]}"
time="2024-02-15T08:32:56-08:00" level=debug msg="iterating vdisk values, data: {Name:storage_vdisk_raidlevel Value:1 Labels:map[controller_name:PERC H730 Mini (Embedded) vdisk:0 vdisk_name:Virtual Disk0]}" vdisk="{storage_vdisk_raidlevel 1 map[controller_name:PERC H730 Mini (Embedded) vdisk:0 vdisk_name:Virtual Disk0]}"
time="2024-02-15T08:32:56-08:00" level=debug msg="iterating vdisk values, data: {Name:storage_vdisk_read_policy Value: Labels:map[controller_name:PERC H730 Mini (Embedded) vdisk:0 vdisk_name:Virtual Disk0]}" vdisk="{storage_vdisk_read_policy  map[controller_name:PERC H730 Mini (Embedded) vdisk:0 vdisk_name:Virtual Disk0]}"
time="2024-02-15T08:32:56-08:00" level=error msg="storage_vdisk collector failed after 0.137535s: strconv.ParseFloat: parsing \"\": invalid syntax"
`

galexrt, 2024-02-15:
Yep, there seem to be "new" (?) vdisk read/write policies that I didn't have in the "mappings" yet. I have just released `v1.13.7`, which contains fixes for these new policies and uses a newer Golang version.

Please try again and let me know if that fixes the vdisk issues as well.

AyoubH, 2024-02-15:
@galexrt just tried v1.13.7, it seems that it did not fix the issue, here is the output

```shell
DEBU[0012] iterating vdisk values, data: {Name:storage_vdisk_status Value:0 Labels:map[controller_name:BOSS-S1 (Embedded) vdisk:0 vdisk_name:VD_R1_1]}  vdisk="{storage_vdisk_status 0 map[controller_name:BOSS-S1 (Embedded) vdisk:0 vdisk_name:VD_R1_1]}"
DEBU[0012] iterating vdisk values, data: {Name:storage_vdisk_state Value:1 Labels:map[controller_name:BOSS-S1 (Embedded) vdisk:0 vdisk_name:VD_R1_1]}  vdisk="{storage_vdisk_state 1 map[controller_name:BOSS-S1 (Embedded) vdisk:0 vdisk_name:VD_R1_1]}"
DEBU[0012] iterating vdisk values, data: {Name:storage_vdisk_raidlevel Value:111 Labels:map[controller_name:BOSS-S1 (Embedded) vdisk:0 vdisk_name:VD_R1_1]}  vdisk="{storage_vdisk_raidlevel 111 map[controller_name:BOSS-S1 (Embedded) vdisk:0 vdisk_name:VD_R1_1]}"
DEBU[0012] iterating vdisk values, data: {Name:storage_vdisk_read_policy Value: Labels:map[controller_name:BOSS-S1 (Embedded) vdisk:0 vdisk_name:VD_R1_1]}  vdisk="{storage_vdisk_read_policy  map[controller_name:BOSS-S1 (Embedded) vdisk:0 vdisk_name:VD_R1_1]}"
ERRO[0012] storage_vdisk collector failed after 0.212522s: strconv.ParseFloat: parsing "": invalid syntax
```

galexrt, 2024-02-15:
@AyoubH Please post the output of `omreport storage vdisk -fmt ssv`

lchopfpt, 2024-02-15:
THe vdisk parser worked for me with version 1.13.7 on this node.

``` time="2024-02-15T11:12:11-08:00" level=debug msg="iterating vdisk values, data: {Name:storage_vdisk_status Value:0 Labels:map[controller_name:PERC H730 Mini (Embedded) vdisk:0 vdisk_name:Virtual Disk0]}" vdisk="{storage_vdisk_status 0 map[controller_name:PERC H730 Mini (Embedded) vdisk:0 vdisk_name:Virtual Disk0]}"
time="2024-02-15T11:12:11-08:00" level=debug msg="iterating vdisk values, data: {Name:storage_vdisk_state Value:1 Labels:map[controller_name:PERC H730 Mini (Embedded) vdisk:0 vdisk_name:Virtual Disk0]}" vdisk="{storage_vdisk_state 1 map[controller_name:PERC H730 Mini (Embedded) vdisk:0 vdisk_name:Virtual Disk0]}"
time="2024-02-15T11:12:11-08:00" level=debug msg="iterating vdisk values, data: {Name:storage_vdisk_raidlevel Value:1 Labels:map[controller_name:PERC H730 Mini (Embedded) vdisk:0 vdisk_name:Virtual Disk0]}" vdisk="{storage_vdisk_raidlevel 1 map[controller_name:PERC H730 Mini (Embedded) vdisk:0 vdisk_name:Virtual Disk0]}"
time="2024-02-15T11:12:11-08:00" level=debug msg="iterating vdisk values, data: {Name:storage_vdisk_read_policy Value:5 Labels:map[controller_name:PERC H730 Mini (Embedded) vdisk:0 vdisk_name:Virtual Disk0]}" vdisk="{storage_vdisk_read_policy 5 map[controller_name:PERC H730 Mini (Embedded) vdisk:0 vdisk_name:Virtual Disk0]}"
time="2024-02-15T11:12:11-08:00" level=debug msg="iterating vdisk values, data: {Name:storage_vdisk_write_policy Value:7 Labels:map[controller_name:PERC H730 Mini (Embedded) vdisk:0 vdisk_name:Virtual Disk0]}" vdisk="{storage_vdisk_write_policy 7 map[controller_name:PERC H730 Mini (Embedded) vdisk:0 vdisk_name:Virtual Disk0]}"
time="2024-02-15T11:12:11-08:00" level=debug msg="iterating vdisk values, data: {Name:storage_vdisk_cache_policy Value:0 Labels:map[controller_name:PERC H730 Mini (Embedded) vdisk:0 vdisk_name:Virtual Disk0]}" vdisk="{storage_vdisk_cache_policy 0 map[controller_name:PERC H730 Mini (Embedded) vdisk:0 vdisk_name:Virtual Disk0]}"
time="2024-02-15T11:12:11-08:00" level=debug msg="storage_vdisk collector succeeded after 0.165204s."
```

AyoubH, 2024-02-16:
@galexrt here is the output of `omreport storage vdisk -fmt ssv`

```shell
List of Virtual Disks in the System

ID;Status;Name;State;Encrypted;Layout;Size;T10 Protection Information Status;Associated Fluid Cache State ;Device Name;Bus Protocol;Media;Read Policy;Write Policy;Cache Policy;Stripe Element Size;Disk Cache Policy

Controller BOSS-S1 (Embedded)

ID;Status;Name;State;Encrypted;Layout;Size;T10 Protection Information Status;Associated Fluid Cache State ;Device Name;Bus Protocol;Media;Read Policy;Write Policy;Cache Policy;Stripe Element Size;Disk Cache Policy
0;Ok;VD_R1_1;Ready;Not Applicable;RAID-1;111.73 GB (119966990336 bytes);No;Not Applicable;Not Available;SATA;SSD;No Read Ahead;Write Through;Not Applicable;64 KB;Disabled
```

galexrt, 2024-02-16:
@AyoubH @lchopfpt  Can you two please try with v1.13.8 to see if the vdisk collector is fixed/ continues to work with the fix I added.

AyoubH, 2024-02-16:
@galexrt it works fine, thank you so much

AyoubH, 2024-02-16:
@galexrt it works fine, thank you
`DEBU[0056] storage_vdisk collector succeeded after 0.208292s`

lchopfpt, 2024-02-16:
vdisk still looking good for me too.  Thanks!

galexrt, 2024-02-17:
@AyoubH @lchopfpt From your messages, it seems that the latest version (`v1.13.8`) fixes both issues mentioned in this thread, right?

If that resolves the issues, I'll go ahead and close the ticket for now.

AyoubH, 2024-02-18:
@galexrt 

I installed the v1.13.8 versions in some servers, and I just found out that the pdsik error is still there in one of them

**DEBUG mode**
`ERRO[0024] ps collector failed after 0.089713s: failed to execute command. exit status 255
ERRO[0024] fans collector failed after 0.093299s: failed to execute command. exit status 255
ERRO[0024] storage_pdisk collector failed after 0.470633s: failed to execute command. exit status 255`

the omreport shows that there is no disk, although it has two, which I can confirm through iDRAC UI

`List of Physical Disks on Controller PCIe SSD Subsystem (Not Available)

Controller PCIe SSD Subsystem (Not Available)
No Physical Disks found`

**Fans and PS collectors**
For this one, It occurs in only very few servers, but I can open another discussion for it
the omreport command says also that there is no fans or Powersupply
`Error! No fan probes found on this system.`
`Error! No instrumented power supplies found on this system.`

galexrt, 2024-02-18:
@AyoubH Please create a new issue for these. As that seems to be mostly about handling the "no components found" exit code for certain cases it seems.

closed 2024-02-20 by galexrt