galexrt/dellhw_exporter #94: dell_hw_nic_status metric returning 1 for functional interfaces

B0go, 2024-02-01:
Hi @galexrt, thanks for this great contribution to the community. Such an exporter is essential for Dell based  on-prem observability! 

While analyzing the values returned by the exporter from a couple of different servers, I noticed that some metrics were returning 1s for network interfaces that were known to be functional. After digging into the code, I [found](https://github.com/galexrt/dellhw_exporter/blob/main/pkg/omreport/omreport.go#L454) that it only marks an interface as functional (0) when the report either returns "Connected" or "Full" in the fourth field. While this covers most scenarios, some edge cases are not covered.

Here is an example of a Team Interface that currently shows 1 in the metric but is functional:

```
Index             : 1
Interface Name    : br0
Vendor            : Linux
Description       : Network Bridge
Redundancy Status : Not Applicable
```

I believe this happens because "Not Applicable" is not covered by the exporter logic. It would be a good improvement to check the field name before matching its value.

I can open a Pull Request including this scenario!

galexrt, 2024-02-02:
@B0go A pull request to fix this would be appreciated. Please add a test for this case as well, thanks!

Let me know if you need any guidance.

B0go, 2024-02-02:
@galexrt I will work on that whenever possible, thanks.

closed 2024-02-06 by galexrt

galexrt, 2024-02-06:
I have just released `v1.13.2` containing your PR, see https://github.com/galexrt/dellhw_exporter/releases/tag/v1.13.2.

Please let me know if your fix works as expected.

B0go, 2024-02-07:
Confirmed!
[image]