galexrt/dellhw_exporter #115: storage_vdisk parsing error

nexix, 2024-05-17:
Seeing a vdisk parsing error on a "Critical" disk

```
Redirecting to /bin/systemctl status dell_hw_exporter.service
● dell_hw_exporter.service - Prometheus exporter for Dell hardware -- Version true
     Loaded: loaded (/etc/systemd/system/dell_hw_exporter.service; enabled; preset: disabled)
     Active: active (running) since Fri 2024-05-17 18:20:58 UTC; 13s ago
   Main PID: 2045713 (dellhw_exporter)
      Tasks: 26 (limit: 100305)
     Memory: 20.0M
        CPU: 5.726s
     CGroup: /system.slice/dell_hw_exporter.service
             └─2045713 /usr/bin/dellhw_exporter

May 17 18:25:41 m0081489.ppops.net dellhw_exporter[2045713]: time="2024-05-17T18:25:41Z" level=error msg="storage_vdisk collector failed after 0.139247s: strconv.ParseFloat: parsing \"\": invalid syntax"
May 17 18:26:07 m0081489.ppops.net dellhw_exporter[2045713]: time="2024-05-17T18:26:07Z" level=error msg="storage_vdisk collector failed after 0.166057s: strconv.ParseFloat: parsing \"\": invalid syntax"
May 17 18:26:11 m0081489.ppops.net dellhw_exporter[2045713]: time="2024-05-17T18:26:11Z" level=error msg="storage_vdisk collector failed after 0.143599s: strconv.ParseFloat: parsing \"\": invalid syntax"
May 17 18:26:37 m0081489.ppops.net dellhw_exporter[2045713]: time="2024-05-17T18:26:37Z" level=error msg="storage_vdisk collector failed after 0.151444s: strconv.ParseFloat: parsing \"\": invalid syntax"
May 17 18:26:41 m0081489.ppops.net dellhw_exporter[2045713]: time="2024-05-17T18:26:41Z" level=error msg="storage_vdisk collector failed after 0.148115s: strconv.ParseFloat: parsing \"\": invalid syntax"
```
```
List of Virtual Disks in the System

Controller PERC H710 Mini (Embedded)

ID;Status;Name;State;Hot Spare Policy violated;Virtual Disk Bad Blocks;Encrypted;Layout;Size;T10 Protection Information Status;Associated Fluid Cache State ;Device Name;Bus Protocol;Media;Read Policy;Write Policy;Cache Policy;Strip Element Size;Disk Cache Policy
0;Ok;Virtual Disk 0;Ready;Not Assigned;No;No;RAID-6;93.13 GiB (99999547392 bytes);No;Not Applicable;/dev/sda;SATA;HDD;Adaptive Read Ahead;Write Back;Not Applicable;128 KB;Enabled
1;Critical;Virtual Disk 1;Ready;Not Assigned;Yes;No;RAID-6;22,259.87 GiB (23901351444480 bytes);No;Not Applicable;/dev/sdb;SATA;HDD;Adaptive Read Ahead;Write Back;Not Applicable;128 KB;Enabled
```
```
dellhw_exporter, version 1.13.12 (branch: HEAD, revision: b8900ff8addca32021e9bbb97258cebe7b0c230c)
  build user:       root@d49947862d6f
  build date:       20240516-20:45:11
  go version:       go1.22.3
  platform:         linux/amd64
  tags:             netgo static_build
```

galexrt, 2024-06-14:
Could you please provide the output of the following command:

```
omreport storage vdisk -fmt ssv
omreport storage vdisk -fmt json
```

(The `json` output as well, as I would like to see how "good" the JSOn output looks like for a potential rewrite to use the json output instead of semicolon separated output from omreport)

fganee-okn, 2024-07-17:
Hi!
Allow me to jump in as we face the same error :D 

```
[root@dellhw-exporter-wnz48 /]# omreport storage vdisk -fmt ssv
List of Virtual Disks in the System

Controller PERC H740P Adapter  (Slot 6)

ID;Status;Name;State;Hot Spare Policy violated;Virtual Disk Bad Blocks;Encrypted;Layout;Size;T10 Protection Information Status;Associated Fluid Cache State ;Device Name;Bus Protocol;Media;Read Policy;Write Policy;Cache Policy;Strip Element Size;Disk Cache Policy
0;Ok;DATA;Ready;Not Applicable;No;No;RAID-0;7,153.38 GiB (7680877920256 bytes);No;Not Applicable;/dev/sda;SAS;SSD;Read Ahead;Write Back;Not Applicable;512 KB;Unchanged
1;Critical;STORAGE;Ready;Not Assigned;Yes;No;RAID-5;49,170.00 GiB (52795885486080 bytes);No;Not Applicable;/dev/sdb;SAS;HDD;Read Ahead;Write Back;Not Applicable;512 KB;Unchanged

ID;Status;Name;State;Hot Spare Policy violated;Virtual Disk Bad Blocks;Encrypted;Layout;Size;T10 Protection Information Status;Associated Fluid Cache State ;Device Name;Bus Protocol;Media;Read Policy;Write Policy;Cache Policy;Strip Element Size;Disk Cache Policy

[root@dellhw-exporter-wnz48 /]# omreport storage vdisk -fmt json
Error! Invalid format for output results specified: json  
```

galexrt, 2024-07-19:
@fganee-okn Thanks! What version of OpenManage/omreport are you running?

galexrt, 2024-10-19:
I will soon release the first release candidate that contains the improved `ssv` parsing and adds the output from @fganee-okn to the test set which seems to work fine with the empty second list/table header.

commit 61cfd577 (2024-10-19): fix: add test case for vdisk parsing issue

Resolves #115
closed 2024-10-19 by galexrt