galexrt/dellhw_exporter #118: dellhw_exporter sets iDrac Hostname, OS and OS version to those of container

cgrijs, 2024-06-11:
Hi,

We are running dellhw_exporter from the helm chart (v1.13.10) and we see that in the iDrac, the Host Name, Operating System and Operating System Version now reflect on what is running in the pod (dellhw-exporter-xxxxx; Rocky Linux; release 9.4 (Blue Onyx) Kernel 5.15.0-112-generic (x86_64)) instead of what is running on the hardware.

galexrt, 2024-06-14:
Could you please post an example metrics output where the pod info is shown instead of the hosts info?

matanbaruch, 2024-11-24:
> Hi,
> 
> We are running dellhw_exporter from the helm chart (v0.2.12) and we see that in the iDrac, the Host Name, Operating System and Operating System Version now reflect on what is running in the pod (dellhw-exporter-xxxxx; Rocky Linux; release 9.4 (Blue Onyx) Kernel 5.15.0-112-generic (x86_64)) instead of what is running on the hardware.

> Could you please post an example metrics output where the pod info is shown instead of the hosts info?

I think he means this

[image]

matanbaruch, 2025-07-02:
@galexrt will check it out!

closed 2026-09-01 by galexrt