galexrt/dellhw_exporter #121: Running v1.13.12 against iSM

p3lim, 2024-07-08:
Hey,

Nice little exporter, tried running it against iSM 5.3.0 / iSM OSC 7.3.0 on iDRAC 9 v7.00.00.00, and here are my findings:

- immediately got an error when starting:
   - `/usr/libexec/instsvcdrv-helper: line 519: lsmod: command not found` x5
   - exporter still starts
- `dell_hw_chassis_temps` all report `0`, but `dell_hw_chassis_temps_reading` have values, unsure what the difference between these two are

All other metrics seems to work fine! :+1:

galexrt, 2024-09-10:
The `lsmod: command not found` can be ignored, as the Dell OMS service startup script tries to check if the necessary modules are there/need to `modprobe`d.

I'll take a look at the `dell_hw_chassis_temps` metric reporting `0` soon.

galexrt, 2024-10-17:
@eugene-marchanka The `dell_hw_chassis_temps` metric is reporting the "status" as a number, e.g., `0` = "OK", see for a section that explains the status numbers https://github.com/galexrt/dellhw_exporter/blob/main/docs/metrics.md#what-do-the-metrics-mean
(The docs site is currently broken which I'm looking into soon)

The `chassis_temps_reading` is the one that reports the actual temperature reading.

eugene-marchanka, 2024-10-17:
> @eugene-marchanka The `dell_hw_chassis_temps` metric is reporting the "status" as a number, e.g., `0` = "OK", see for a section that explains the status numbers [...]

Thanks @galexrt !
I successfully installed `dellhw_exporter` to our qa/dev environments and ready to roll it out to prod systems 👍🏻

galexrt, 2024-10-17:
Great to hear!

I think as the issue itself stands, the questions/points have been answered so I'm going to close the issue.

I might revisit the issue in the future to add the `dell_hw_chassis_temps` question to a FAQ section.

closed 2024-10-17 by galexrt