galexrt/dellhw_exporter #124: ignoring exit code 255 omreports can lose error cases

juliantaylor, 2024-08-22:
omreport commands that fail with exit code 255 are ignored which was done in https://github.com/galexrt/dellhw_exporter/issues/99

in this case the no metrics for the components are created at all so monitoring of data can be silently lost unless you have absent() rules for all metrics being alerted on.

In our experience the dell tools are somewhat fragile and reports of components may sometimes just fail until the machine is rebooted or moduls reloaded (frequently after firmware updates), so it would be better to have some way of configuring which components you expect to fail due to them not being present and components you expect to be there and have this represented in the dell_hw_scrape_collector_success for simple alerting.

For example following command fails with 255 due to no components found which is ok as the disks are all on the perc raidcontroller component (an option to ignore controllers may be useful here or skip controllers which report zero in`Occupied Slot Count` of info=pdslotreport)

```
omreport storage pdisk controller=1
Controller PCIe SSD Subsystem (Not Available)
No Physical Disks found
```

But e.g. this command fails with 255 due to an error:

```
# omreport chassis batteries
Error! No battery probes found on this system.
```

the battery probes are present there is just an issue in exporting them due to a certain buggy firmware version.

sfudeus, 2024-08-26:
To solve this and !99, I suggest to not just ignore the errors, but configure which components to monitor at all (should address this blade issue from !99) or which to process but ignore errors not finding them. 
Regarding the case of the raid controllers, of course something slightly more intelligent is required, because we DO want to monitor them, but a raid controller of course have no disks attached. 

Examples:

with disks:
```
omreport storage controller controller=0 info=pdslotreport -fmt ssv; echo "status $?"

Details of Enclosure BP_EXP           on Controller PERC H355 Front

ID;Slot Count ;Occupied Slot Count ;Empty Slot Count ;Occupied Slots ;Empty Slots
0:1;24;2;22;0,1;2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23

Occupied Physical Disk Details

ID;Status;Name;State;Power Status;Bus Protocol;Media;Part of Cache Pool;Remaining Rated Write Endurance;Failure Predicted;Revision;Driver Version;Model Number;T10 PI Capable;Certified;Encryption Capable;Encryption Protocol;Encrypted;Progress;Mirror Set ID;Capacity;Used RAID Disk Space;Available RAID Disk Space;Hot Spare;Vendor ID;Product ID;Serial No.;Part Number;Negotiated Speed;Capable Speed;PCIe Negotiated Link Width;PCIe Maximum Link Width;Logical Block size;Device Write Cache;Manufacture Day;Manufacture Week;Manufacture Year;SAS Address;WWN;Non-RAID HDD Disk Cache Policy;Disk Cache Policy;Sub Vendor;Cryptographic Erase Capable
0:1:0;Ok;Physical Disk 0:1:0;Non-RAID;Not Applicable;SATA;SSD;Not Applicable;100%;No;D4DK003;Not Applicable;Not Applicable;No;Yes;No;Not Applicable;Not Applicable;Not Applicable;Not Applicable;446.63 GiB (479559942144 bytes);0.00 GiB (0 bytes);446.63 GiB (479559942144 bytes);No;DELL(tm);MTFDDAK480TGB-1BC1ZABDA;24084747D28D;SG0R6XYWMCS0043325P3A00;6.00 Gbps;6.00 Gbps;Not Applicable;Not Applicable;512B;Not Applicable;Not Available;Not Available;Not Available;0x500056B3277C5DC0;0x500056B3277C5DC0;Not Applicable;Not Applicable;Not Available;Yes
0:1:1;Ok;Physical Disk 0:1:1;Non-RAID;Not Applicable;SATA;SSD;Not Applicable;100%;No;D4DK003;Not Applicable;Not Applicable;No;Yes;No;Not Applicable;Not Applicable;Not Applicable;Not Applicable;446.63 GiB (479559942144 bytes);0.00 GiB (0 bytes);446.63 GiB (479559942144 bytes);No;DELL(tm);MTFDDAK480TGB-1BC1ZABDA;24084747D35A;SG0R6XYWMCS0043325P4A00;6.00 Gbps;6.00 Gbps;Not Applicable;Not Applicable;512B;Not Applicable;Not Available;Not Available;Not Available;0x500056B3277C5DC1;0x500056B3277C5DC1;Not Applicable;Not Applicable;Not Available;Yes

status 0
```

without disks:
```
omreport storage controller controller=1 info=pdslotreport -fmt ssv; echo "status $?"

Details of Enclosure Backplane1 on Controller PCIe SSD Subsystem

ID;Slot Count ;Occupied Slot Count ;Empty Slot Count ;Occupied Slots ;Empty Slots
0:1;8;0;8;Not Available;16,17,18,19,20,21,22,23

ID;Status;Name;State;Power Status;Bus Protocol;Media;Part of Cache Pool;Remaining Rated Write Endurance;Failure Predicted;Revision;Driver Version;Model Number;T10 PI Capable;Certified;Encryption Capable;Encryption Protocol;Encrypted;Progress;Mirror Set ID;Capacity;Used RAID Disk Space;Available RAID Disk Space;Hot Spare;Vendor ID;Product ID;Serial No.;Part Number;Negotiated Speed;Capable Speed;PCIe Negotiated Link Width;PCIe Maximum Link Width;Logical Block size;Device Write Cache;Manufacture Day;Manufacture Week;Manufacture Year;SAS Address;WWN;Non-RAID HDD Disk Cache Policy;Disk Cache Policy;Form Factor ;Sub Vendor;Cryptographic Erase Capable

status 0
```

Note that the controller report exits with 0, but the pdisk report exists with 255.

sfudeus, 2024-08-26:
Though there already is `--collectors-enabled`, so people could already disable irrelevant (i.e. those throwing errors for a cause) collectors. Suggesting to fix this raid controller issue more specifically than with ignoring all errors.

juliantaylor, 2024-08-30:
`storage controller controller=%d info=pdslotreport` reports the same data as `storage controller pdisk controller=%d` and works on empty controllers, so this patch could work:

```patch
--- a/pkg/omreport/omreport.go
+++ b/pkg/omreport/omreport.go
@@ -247,11 +247,15 @@ func (or *OMReport) StorageEnclosure() ([]Value, error) {
 func (or *OMReport) StoragePdisk(cid string) ([]Value, error) {
        values := []Value{}
        controllerName := "N/A"
+       pdisk_data := false
        err := or.readReport(func(fields []string) {
-               if len(fields) == 1 && strings.HasPrefix(fields[0], omReportControllerNamePrefix) {
-                       controllerName = strings.TrimPrefix(fields[0], omReportControllerNamePrefix)
+               if len(fields) == 1 && strings.Contains(fields[0], "on Controller") {
+                       controllerName = strings.TrimSpace(strings.Split(fields[0], "on Controller")[1])
                        return
-               } else if len(fields) < 3 || fields[0] == "ID" {
+               } else if len(fields) == 1 && fields[0] == "Occupied Physical Disk Details" {
+                       pdisk_data = true
+                       return
+               } else if len(fields) < 3 || fields[0] == "ID" || !pdisk_data {
                        return
                }
                // Need to find out what the various ID formats might be
@@ -308,7 +312,7 @@ func (or *OMReport) StoragePdisk(cid string) ([]Value, error) {
                                })
                        }
                }
-       }, or.getOMReportExecutable(), "storage", "pdisk", "controller="+cid)
+       }, or.getOMReportExecutable(), "storage", "controller", "controller="+cid, "info=pdslotreport")
        return values, err
 }
 
diff --git a/pkg/omreport/omreport_test.go b/pkg/omreport/omreport_test.go
index e2178f0..dfcab8f 100644
--- a/pkg/omreport/omreport_test.go
+++ b/pkg/omreport/omreport_test.go
@@ -504,9 +504,12 @@ func TestStorageEnclosure(t *testing.T) {
 
 var storagePdiskTests = []testResultOMReport{
        {
-               Input: `List of Physical Disks on Controller PERC H730 Mini (Slot Embedded)
+               Input: `Details of Enclosure BP13G_EXP        on Controller PERC H730 Mini (Slot Embedded)
 
-Controller PERC H730 Mini (Slot Embedded)
+ID;Slot Count ;Occupied Slot Count ;Empty Slot Count ;Occupied Slots ;Empty Slots
+0:1;10;3;7;0,1,2;3,4,5,6,7,8,9
+
+Occupied Physical Disk Details
 
 ID;Status;Name;State;Power Status;Bus Protocol;Media;Part of Cache Pool;Remaining Rated Write Endurance;Failure Predicted;Revision;Driver Version;Model Number;T10 PI Capable;Certified;Encryption Capable;Encrypted;Progress;Mirror Set ID;Capacity;Used RAID Disk Space;Available RAID Disk Space;Hot Spare;Vendor ID;Product ID;Serial No.;Part Number;Negotiated Speed;Capable Speed;PCIe Negotiated Link Width;PCIe Maximum Link Width;Sector Size;Device Write Cache;Manufacture Day;Manufacture Week;Manufacture Year;SAS Address;Non-RAID HDD Disk Cache Policy;Disk Cache Policy;Form Factor ;Sub Vendor;ISE Capable
 0:1:0;Ok;Physical Disk 0:1:0;Ready;Not Applicable;SATA;SSD;Not Applicable;100%;No;G201DL2B;Not Applicable;Not Applicable;No;Yes;No;Not Applicable;Not Applicable;Not Applicable;185.75 GB (199447543808 bytes);185.75 GB (199447543808 bytes);0.00 GB (0 bytes);Dedicated;DELL(tm);INTEL SSDSC2BX200G4R;BTHC643503A2200TGN;CN03481GIT2006AT00P3A0;6.00 Gbps;6.00 Gbps;Not Applicable;Not Applicable;512B;Not Applicable;Not Available;Not Available;Not Available;500056B3B43B8CC0;Not Applicable;Not Applicable;Not Available;Not Available;No
@@ -626,7 +629,12 @@ ID;Status;Name;State;Power Status;Bus Protocol;Media;Part of Cache Pool;Remainin
        },
        {
                // State: "Non-RAID" - https://github.com/galexrt/dellhw_exporter/issues/93#issuecomment-1945674675
-               Input: `List of Physical Disks on Controller PERC H330 Mini (Embedded)
+               Input: `Details of Enclosure BP13G_EXP        on Controller PERC H330 Mini (Embedded)
+
+ID;Slot Count ;Occupied Slot Count ;Empty Slot Count ;Occupied Slots ;Empty Slots
+0:1;10;2;8;0,1;2,3,4,5,6,7,8,9
+
+Occupied Physical Disk Details
 
 Controller PERC H330 Mini (Embedded)
 
@@ -672,6 +680,17 @@ ID;Status;Name;State;Power Status;Bus Protocol;Media;Part of Cache Pool;Remainin
                        },
                },
        },
+       {
+               // empty slots - https://github.com/galexrt/dellhw_exporter/issues/99 https://github.com/galexrt/dellhw_exporter/issues/124
+               Input: `Details of Enclosure Backplane1 on Controller PCIe SSD Subsystem
+
+ID;Slot Count ;Occupied Slot Count ;Empty Slot Count ;Occupied Slots ;Empty Slots
+0:1;8;0;8;Not Available;16,17,18,19,20,21,22,23
+
+ID;Status;Name;State;Power Status;Bus Protocol;Media;Part of Cache Pool;Remaining Rated Write Endurance;Failure Predicted;Revision;Driver Version;Model Number;T10 PI Capable;Certified;Encryption Capable;Encryption Protocol;Encrypted;Progress;Mirror Set ID;Capacity;Used RAID Disk Space;Available RAID Disk Space;Hot Spare;Vendor ID;Product ID;Serial No.;Part Number;Negotiated Speed;Capable Speed;PCIe Negotiated Link Width;PCIe Maximum Link Width;Logical Block size;Device Write Cache;Manufacture Day;Manufacture Week;Manufacture Year;SAS Address;WWN;Non-RAID HDD Disk Cache Policy;Disk Cache Policy;Form Factor ;Sub Vendor;Cryptographic Erase Capable
+`,
+               Values: []Value{},
+       },
 }
 
 func TestStoragePdisk(t *testing.T) {

```

galexrt, 2024-10-18:
Thanks! I'll take a look at the proposed patch soon, especially in light of #134 in which the whole parsing isn't dependent in fields by index anymore making fixing the issue probably even easier than having to check the fields length as before. :wink:

juliantaylor, 2024-10-18:
we may need to revise the patch a bit, we just got a perc h9xx and info=pdslotreport does not work on that one ...
its an error instead of unsupported operation, so still need to check if its something dell may fix.

juliantaylor, 2024-11-21:
actually info=pdslotreport is not even needed, the controller information contains the physical disk data already so this works on all our currently in use hardware generations.

I still have to adjust it to the new parsing in 2.x though.

```
--- a/pkg/omreport/omreport.go
+++ b/pkg/omreport/omreport.go
@@ -247,11 +247,18 @@ func (or *OMReport) StorageEnclosure() ([]Value, error) {
 func (or *OMReport) StoragePdisk(cid string) ([]Value, error) {
        values := []Value{}
        controllerName := "N/A"
+       pdisk_data := false
        err := or.readReport(func(fields []string) {
                if len(fields) == 1 && strings.HasPrefix(fields[0], omReportControllerNamePrefix) {
                        controllerName = strings.TrimPrefix(fields[0], omReportControllerNamePrefix)
                        return
-               } else if len(fields) < 3 || fields[0] == "ID" {
+               } else if len(fields) == 1 && fields[0] == "Physical Disks" {
+                       pdisk_data = true
+                       return
+               } else if len(fields) == 1 && fields[0] != "" && fields[0] != "Physical Disks" {
+                       pdisk_data = false
+                       return
+               } else if len(fields) < 3 || fields[0] == "ID" || !pdisk_data {
                        return
                }
                // Need to find out what the various ID formats might be
@@ -308,7 +315,7 @@ func (or *OMReport) StoragePdisk(cid string) ([]Value, error) {
                                })
                        }
                }
-       }, or.getOMReportExecutable(), "storage", "pdisk", "controller="+cid)
+       }, or.getOMReportExecutable(), "storage", "controller", "controller="+cid)
        return values, err
 }
```