galexrt/dellhw_exporter #154: Chart name is not valid

mnaser, 2025-04-23:
The current chart name is not valid based on Helm which is currently `dellhw_exporter`.

https://helm.sh/docs/chart_best_practices/conventions/#chart-names

> Neither uppercase letters nor underscores can be used in chart names. Dots should not be used in chart names.

I would like to propose renaming the chart to `prometheus-dellhw-exporter` to be in line with the rest of the community charts: https://github.com/prometheus-community/helm-charts/tree/main/charts

galexrt, 2025-04-23:
> The current chart name is not valid based on Helm which is currently dellhw_exporter.

_Gotta love how the official helm chart-releaser cli doesn't complain/log a warning about this convention... /s_

Thanks for the PR! Renaming the chart sounds especially to be in line with the rest of the community charts sounds good.

galexrt, 2025-04-23:
PR has been merged and the helm chart has been published https://github.com/galexrt/dellhw_exporter/releases/tag/prometheus-dellhw-exporter-1.0.2 (new version that disables the pod security policies by default).

closed 2025-04-23 by galexrt

mnaser, 2025-04-23:
@galexrt awesome, thanks for the prompt response and the exporter.

larainema, 2025-04-25:
@galexrt seems the 1.0.2 version didn't publish to https://galexrt.github.io/dellhw_exporter
```
helm search repo dellhw
NAME                                      	CHART VERSION	APP VERSION	DESCRIPTION
dellhw_exporter/dellhw_exporter           	1.0.1        	v2.0.0-rc.1	A Helm chart for the dellhw_exporter
dellhw_exporter/prometheus-dellhw-exporter	1.0.1        	v2.0.0-rc.1	A Helm chart for the dellhw_exporter

galexrt, 2025-04-25:
@larainema It seems that the CI runs overlapped and caused the GitHub pages run to fail.

I have released chart version 1.0.3 to ensure that a new version "breaks through any caches" (after `helm repo update`):

```
NAME                                            CHART VERSION   APP VERSION     DESCRIPTION
dellhw_exporter/dellhw_exporter                 1.0.1           v2.0.0-rc.1     A Helm chart for the dellhw_exporter
dellhw_exporter/prometheus-dellhw-exporter      1.0.3           v2.0.0-rc.1     A Helm chart for the dellhw_exporter
```

larainema, 2025-04-25:
@galexrt the 1.0.3 version is available from my side, thanks.